me-no-dev/AsyncTCP #2: Missing dependency?

bofh69, 2017-09-11:
My project's travis build (https://travis-ci.org/bofh69/AiLight/builds/274342483) started to fail today:
```
In file included from .piolibdeps/AsyncTCP_ID1826/src/AsyncTCP.cpp:24:0:
.piolibdeps/AsyncTCP_ID1826/src/AsyncTCP.h:28:29: fatal error: freertos/semphr.h: No such file or directory
#include "freertos/semphr.h"
^
compilation terminated.
```
Shouldn't there be a dependency on the freetos library from this library?

me-no-dev, 2017-09-12:
is this ESP32?

me-no-dev, 2017-09-12:
probably PIO issue that we resolved last night :) pull the latest changes and should be fine

bofh69, 2017-09-12:
Yes, it was with PIO and now it works fine again. :-) Thanks!

closed 2017-09-12 by bofh69

terryjmyers, 2017-10-30:
What's PIO?  I'm having the same issue right now, ESP8266 toolchain.