me-no-dev/AsyncTCP #4: AsyncClient::add() never returns

drejkus, 2017-10-06:
Hi,
I have issue with `AsyncClient::add()` - it never returns, when adding only 2 bytes. 
I'm actually trying to modify https://github.com/marvinroger/async-mqtt-client to work with ESP32, but it fails right at the beginning in the `onConnect` callback where it tries to assemeble the packet. 

Please do you have any clue what goes wrong or where to look?

me-no-dev, 2017-10-08:
Hmm... that is rather strange... need to investigate

drejkus, 2017-10-11:
Just adding simple test file. 

[AsyncTest.zip](https://github.com/me-no-dev/AsyncTCP/files/1375730/AsyncTest.zip)

copercini, 2017-10-13:
I'm hitting the exactly same issue, one of the problems is in dns resolution, AsyncMqttClient doesn't have a `dns_found_callback` function to continue after AsyncTCP return an IP here: https://github.com/me-no-dev/AsyncTCP/blob/master/src/AsyncTCP.cpp#L555

with IP directly, [seems to be something between _tcp_write and _tcp_write_api](https://github.com/me-no-dev/AsyncTCP/blob/master/src/AsyncTCP.cpp#L245) because none of my debug messages in `_tcp_write_api` are printed, and the last one is in `_tcp_write`

me-no-dev, 2017-10-13:
Aaaahhhh... onConnect runs on the network thread... let me try to figure this out...

me-no-dev, 2017-10-13:
guys try the latest commit :)

drejkus, 2017-10-13:
Brilliant! Works for me. 

@copercini - I hit the DNS issue aswell, but I belive that the problem is in AsyncMqttClient where it does not return AsyncTCP `connect()` result in it's own [`connect()`](https://github.com/marvinroger/async-mqtt-client/blob/ddbf4d13287f72bbd281b463f5f032cfe035869f/src/AsyncMqttClient.cpp#L677). If it did we could simply issue a reconnect. But that is another story

me-no-dev, 2017-10-13:
dns issue should also be fixed ;)

copercini, 2017-10-13:
Awesome man!!!

Both problems was solved here, thanks!

me-no-dev, 2017-10-13:
:) closing then :P

closed 2017-10-13 by me-no-dev