me-no-dev/AsyncTCP #5: incompatible with new SDK 2.1

universam1, 2017-10-17:
unable to build with latest commits in Arduino for SDK 2.1 - I conclude this is the reason.

Please see the log output here:
https://travis-ci.org/universam1/iSpindel/builds/288929668

me-no-dev, 2017-10-17:
uhmm this is the ESP32 lib. I fixed the 8266 one last night. Maybe pull the changes?

universam1, 2017-10-17:
Oh sorry for confusing the repos, and thanks for the quick response!

That sounds great, unfortunately PIO is not updating the repo, did you bump the version?

```
pio lib update
Library Storage: C:\src\Dropbox\Arduino\iSpindel\.piolibdeps
Updating ESPAsyncTCP                     @ 1.1.0          [Up-to-date]
Updating ESPAsyncWebServer               @ 1.1.0          [Up-to-date]
```

me-no-dev, 2017-10-17:
have not bumped it because I did not have the time to finish everything (SSL) and I did not test it thuraly

universam1, 2017-10-17:
Since I am using travis with PIO I have no idea how to pull more recent library commits than the releases. If you got an idea, would be glad.

universam1, 2017-10-23:
closing to move to correct repo

closed 2017-10-23 by universam1