me-no-dev/AsyncTCP #7: Unify the ESP32 and ESP8266 version.

BlackEdder, 2017-10-31:
My library works on both ESP32 and ESP8266, but I had trouble getting my project to reliably switch between using AsyncTCP and ESPAsyncTCP depending on the hardware (this was with platformio). In the end I decided to make AsyncTCP also work/compile on ESP8266 hardware. If you are interested the relevant files/changes can be found here:

https://gitlab.com/BlackEdder/painlessMesh/blob/tcp_locking/src/AsyncTCP.h
https://gitlab.com/BlackEdder/painlessMesh/blob/tcp_locking/src/AsyncTCP.cpp

me-no-dev, 2017-10-31:
switching is quite easy ;) it's done by define switch like here: https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/ESPAsyncWebServer.h#L31-L39
Platformio also has ways for that to work (it should with later versions out of the box. also look at the web server lib config).
With what you have above, you first have a copy of this lib and second will miss on development for each platform. It was not worth it keeping them in the same lib because the platforms are vastly different and that will only cause code clutter.
Not to say I do not even see you mention my two libs in your readme ;)

BlackEdder, 2017-10-31:
The problem I keep running into is that when I compile for one board and then the other platformio tries to include both libraries, i.e. I have to delete the temporary directories before I can switch. Maybe I am doing something wrong though, but I'd rather not have users have to deal with that.

> Not to say I do not even see you mention my two libs in your readme ;)

The development is on a merge request currently ;) Will certainly add a reference when/if merged

me-no-dev, 2017-10-31:
@ivankravets have a clue about this? I would rather @BlackEdder not have to use some sort of both libs into one to make it work.

@BlackEdder would be much better if you try to make it work as it should, instead of going the way you went ;)

BlackEdder, 2017-10-31:
@ivankravets You can recreate the issue I mean by running:

```
git clone https://gitlab.com/BlackEdder/painlessMesh.git
cd painlessMesh
pio run -t upload -e nodemcuv2 -d examples/webServer/
pio run -t upload -e esp32 -d examples/webServer/
```

The nodemcuv2 build will work, while the esp32 build will fail, because it tries to include ESPAsyncTCP.cpp

ivankravets, 2017-10-31:
@BlackEdder Just replace `ESPAsyncWebServer` with `AsyncWebServer` in https://gitlab.com/BlackEdder/painlessMesh/blob/master/examples/webServer/platformio.ini#L17 

See

* https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/espressif8266_library.json
* https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/espressif32_library.json

BlackEdder, 2017-10-31:
Sorry, but that still does not make it possible to compile them both in
order, without having to delete the temporary directories.

As far as I can tell this is because platformio tries to include all
libraries in the temporary build directory, without checking dependencies.
A possible solution would be for platformio to have `env` specific build
directories. Which is also why the solution proposed in
https://github.com/me-no-dev/ESPAsyncWebServer/issues/223 did not work:

```
{
  "name":"ESPAsyncWebServer",
  "description":"Asynchronous HTTP and WebSocket Server Library for ESP8266
and ESP32",
  "keywords":"http,async,websocket,webserver",
  "authors":
  {
    "name": "Hristo Gochkov",
    "maintainer": true
  },
  "repository":
  {
    "type": "git",
    "url": "https://github.com/me-no-dev/ESPAsyncWebServer.git"
  },
  "version": "1.1.0",
  "license": "LGPL-3.0",
  "frameworks": "arduino",
  "platforms":"espressif8266, espressif32",
  "dependencies": [
    {
      "name": "ESPAsyncTCP",
      "platforms": "espressif8266"
    },
    {
      "name": "AsyncTCP",
      "platforms": "espressif32"
    }
  ]
}
```

ivankravets, 2017-10-31:
@BlackEdder Thanks! I see an error. I'll take a look soon and reply here. Temporary solution is to put  `ESPAsyncTCP` to ignore list for ESP32 build environment
```
[env:esp32]
...
lib_ignore = ESPAsyncTCP
```

BlackEdder, 2017-11-01:
@ivankravets Thanks!

I guess there is no lib_ignore for library.json? That way the library that depends on either `AsyncTCP` or `ESPAsyncTCP` could lib_ignore the one that causes a conflict :)

andig, 2017-11-02:
I'm seeing similar problems as @BlackEdder with library resolution, see https://github.com/platformio/platformio-core/issues/1111. I've also switched to `lib_ignore` for short-term fix. This inlcudes ignoring `AsyncTCP` on ESP8266, too.

ivankravets, 2017-11-04:
This is a bug and we have an issue for that https://github.com/platformio/platformio-core/issues/1153

I'm going to fix it as soon as possible.

closed 2019-10-05 by stale[bot]

gmag11, 2019-11-19:
I have a similar problem with VisualMicro. When compiling with ESP8266 or ESP32 it tries to include AsyncTCP.h and ESPAsyncTCP.h in both cases. Should I blame VisualMicro?