me-no-dev/AsyncTCP #8: SSL support?

Eroc33, 2017-11-18:
I was under the impression I would be able to use ssl with [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer) on esp32, however as this library doesn't support ssl (whereas it's esp8266 counterpart does) it appears that's not possible?

It would be nice for this lib to be on par with the esp8266 version in terms of features to avoid surprises like this.

Is ssl support a feature that is likely to get implemented?

me-no-dev, 2017-11-18:
yes it is :) as soon as I have time to investigate Async SSL. Not as easy as normal SSL

h3ndrik, 2018-01-29:
basecamp is also waiting for someone to implement ssl support:

https://github.com/merlinschumacher/Basecamp/issues/8

LaurentLouf, 2018-09-28:
Any news on that ?

fremouw, 2018-11-29:
Hi, I've been working lately on adding TLS support using mbed-tls, I've got it working but the implementation is still a bit rough and needs some cleaning and improvements (and some testing). Anyway, feel free to check it out and make improvements, you can find my fork here: https://github.com/fremouw/AsyncTCP/tree/mbed-tls.

mandulaj, 2019-05-09:
Bump
Is there any news on this? I am trying to use SSL on my MQTT connection from an ESP32. Since I am using async-mqtt-client, I have managed to trace down the culprit of my failure to here. As far as I can see, the only option is to use Pubsubclient instead right now.

LaurentLouf, 2019-05-09:
If you can update your esp-idf version, there's an MQTT library included in recent versions. And it supports SSL. From esp-idf 3.3 I think.

fremouw, 2019-05-09:
> Bump
> Is there any news on this? I am trying to use SSL on my MQTT connection from an ESP32. Since I am using async-mqtt-client, I have managed to trace down the culprit of my failure to here. As far as I can see, the only option is to use Pubsubclient instead right now.

I've created a pull request to add support for TLS, see: https://github.com/me-no-dev/AsyncTCP/pull/43. Works only as a client and I've been using it for MQTTS for some time now.

tve, 2019-09-21:
Given #48 I believe this can be closed...

closed 2019-09-22 by me-no-dev

jeroenst, 2019-10-31:
The issue is not resolved (yet) so imo this issue should still be open, right?

tve, 2019-10-31:
It has been superseded by #48

Tanmay-FuelDaddy, 2023-05-17:
So It is concluded that this Library is not going to support Async SSL/TLS , but the ESP8266 gets it? 
Perhaps there is a way to port that Lib to 32.