me-no-dev/AsyncTCP #10: ETH support

heleon19, 2018-01-23:
Can the lib be used with ETH on the ESP32 instead of WiFi?

copercini, 2018-02-28:
yes, I'm using with ethernet

ahk97, 2019-06-27:
Could you post any sample code, please?
I am using Wifi and ETH / W5500 and the AsyncTCP Webserver will only bind to the wifi interface
thanks

me-no-dev, 2019-06-28:
not W5500 :) use a real ethernet PHY :) W5500 works through Arduino and does not attach to the LwIP stack. It will never work with Async.

ahk97, 2019-08-22:
Which PHY would you recommend?
And, if I replace the W5500 with the recommended PHY, will WiFi and ETH work simultaneously with AsyncTCP? (I would like to start one AsyncWebServer and serve requests via either Interface).
Thanks!

atanisoft, 2019-08-22:
@ahk97 any that are supported by esp-idf. The example readme for Ethernet phy with IDF has a few types that are supported: https://github.com/espressif/esp-idf/blob/master/examples/ethernet/basic/README.md

closed 2019-11-04 by stale[bot]