me-no-dev/AsyncTCP #11: esp32 error while serving files from SPIFFS

lefedor, 2018-02-26:
Hello, I have error with AsyncTCP while serving files from SPIFFS:

Guru Meditation Error: Core  1 panic'ed (StoreProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x4013ad4f  PS      : 0x00060130  A0      : 0x800d5311  A1      : 0x3ffe2340  
A2      : 0xdceb8fcd  A3      : 0x00000000  A4      : 0x00060120  A5      : 0x3ffe23e0  
A6      : 0x3ffdfd2c  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffe2350  
A10     : 0x3ffca6e0  A11     : 0x3ffdfcfc  A12     : 0x000022c2  A13     : 0x3ffe3da8  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000019  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xdceb8ffd  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4013ad4f:0x3ffe2340 0x400d530e:0x3ffe2360 0x400d55ed:0x3ffe23a0 0x400d565b:0x3ffe23d0

Which decodes to:

Decoding 7 results
0x4013ad4f: tcp_arg at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/tcp.c line 1652
0x4013ad4f: tcp_arg at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/tcp.c line 1652
0x400d530e: AsyncClient::_close() at /home/lefedor/Arduino/libraries/AsyncTCP.new/src/AsyncTCP.cpp line 787
0x400d55ed: AsyncClient::_poll(tcp_pcb*) at /home/lefedor/Arduino/libraries/AsyncTCP.new/src/AsyncTCP.cpp line 787
0x400d565b: AsyncClient::_s_poll(void*, tcp_pcb*) at /home/lefedor/Arduino/libraries/AsyncTCP.new/src/AsyncTCP.cpp line 787
:  (inlined by) _handle_async_event at /home/lefedor/Arduino/libraries/AsyncTCP.new/src/AsyncTCP.cpp line 84
:  (inlined by) _async_service_task at /home/lefedor/Arduino/libraries/AsyncTCP.new/src/AsyncTCP.cpp line 96

Have tried current and prev git versions AsyncTCP && current and prev version of arduino-esp32 - results are the same. Any Idea how to fix it?

lefedor, 2018-02-26:
I do also have another reboot event happens, don't know if it goes together with first one:

ssertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./heap_caps.c", line 274, function: heap_caps_free

Backtrace: 0x40087f48:0x3ffd3f40 0x40088047:0x3ffd3f60 0x400eb213:0x3ffd3f80 0x40083d67:0x3ffd3fb0 0x400842f9:0x3ffd3fd0 0x4000bec7:0x3ffd3ff0 0x40109859:0x3ffd4010 0x4010991b:0x3ffd4030 0x400d55b1:0x3ffd4050 0x40105f95:0x3ffd4070

Rebooting...
ets Jun  8 2016 00:22:57

which translates to :

0x40087f48: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 578
0x40088047: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 578
0x400eb213: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c line 63 (discriminator 8)
0x40083d67: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./heap_caps.c line 136
0x400842f9: _free_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/./syscalls.c line 42
0x40109859: tcp_close_shutdown at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/tcp.c line 225
0x4010991b: tcp_close at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/tcp.c line 305
0x400d55b1: _tcp_close_api(tcpip_api_call*) at /home/lefedor/Arduino/libraries/AsyncTCP.new/src/AsyncTCP.cpp line 787
0x40105f95: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/tcpip.c line 474

copercini, 2018-03-07:
I'm also getting it sometimes

fcouceiro, 2018-04-03:
I'm also experiencing this. I tried to check heap integrity around the poll callback using `heap_caps_check_integrity(MALLOC_CAP_8BIT, true)`, but I don't think it is an isolated case. 
More like a timing issue, as it only occurs when I make requests very quickly and with varying stacktraces.

Different backtraces: [#1](https://pastebin.com/raw/BLbkFTHZ), [#2](https://pastebin.com/raw/G3crrkRj), [#3](https://pastebin.com/raw/a4PyBjSh)

CelliesProjects, 2018-05-03:
I have it happen too:
````
CORRUPT HEAP: Bad head at 0x3ffe04e4. Expected 0xabba1234 got 0x3ffe0668
assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./multi_heap_poisoning.c", line 201, function: multi_heap_free
abort() was called at PC 0x400faf33 on core 1

Backtrace: 0x400919e8:0x3ffdef50 0x40091bc3:0x3ffdef70 0x400faf33:0x3ffdef90 0x400916f9:0x3ffdefc0 0x400843fa:0x3ffdefe0 0x40089bb5:0x3ffdf000 0x4000bec7:0x3ffdf020 0x40118645:0x3ffdf040 0x40118707:0x3ffdf060 0x400dd85d:0x3ffdf080 0x40114a21:0x3ffdf0a0
````
which decodes to:
````
x400faf33: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c line 63 (discriminator 8)
0x400919e8: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 648
0x40091bc3: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 648
0x400faf33: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c line 63 (discriminator 8)
0x400916f9: multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./multi_heap_poisoning.c line 297
0x400843fa: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./heap_caps.c line 123
0x40089bb5: _free_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/./syscalls.c line 42
0x40118645: tcp_close_shutdown at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/tcp.c line 835
0x40118707: tcp_close at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/tcp.c line 835
0x400dd85d: _tcp_close_api(tcpip_api_call*) at /home/cellie/Arduino/libraries/AsyncTCP-master/src/AsyncTCP.cpp line 702
0x40114a21: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/tcpip.c line 474
````
And the occasional
````
[W][AsyncTCP.cpp:543] _poll(): rx timeout 4
[W][AsyncTCP.cpp:543] _poll(): rx timeout 4
````
In the serial logs.

JohnGENZ, 2018-08-18:
I have a similar error - the web server is very stable except for this very occasional error at times of high load and/or rapid requests for the web page.

Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x4011a36b  PS      : 0x00060c30  A0      : 0x8011664e  A1      : 0x3ffe29f0  
A2      : 0x3fff499c  A3      : 0x3ffd3660  A4      : 0x00000002  A5      : 0x00000000  
A6      : 0x3ffdff34  A7      : 0x00003231  A8      : 0x0000000f  A9      : 0xabba1244  
A10     : 0x00000008  A11     : 0x2e343131  A12     : 0x00000010  A13     : 0xabba1234  
A14     : 0x00000000  A15     : 0x0000111c  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000013  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

decodes to:

PC: 0x4011a36b: tcp_output at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/tcp_out.c line 1007
EXCVADDR: 0x00000013

Decoding stack results
0x4011a36b: tcp_output at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/tcp_out.c line 1007
0x4011664b: tcp_recved at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/tcp.c line 678
0x400ecf49: _tcp_recved_api(tcpip_api_call*) at C:\Users\John\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp line 256
0x4011365d: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/tcpip.c line 127

me-no-dev, 2018-08-19:
with my current code base those errors start to make less and less sense :( I did find and patch some issues in the last few days, but that did not stop some exceptions like the one above... (you have updated to the last code before posting the above right?)

JohnGENZ, 2018-08-19:
Many thanks.  The last update was about 4 days ago and I have been trying to find a consistent way to reproduce the error - I will update again and see how I go.

malbrook, 2018-09-22:
We have a fairly complex application that uses both cores, one for WiFi and one for our application and we use SPIFFS to hold the web pages, js and css files that are served to the client browser. Up until a few days ago we had a very unstable operation of the application with regular exception errors when accessing the web pages via the browser. We are using jquery to provide internal web page upgrades on a timer in the page code so there is a regular communication (once every two to five seconds) between browser and the ESP32 server. 

All the exceptions point into ESPAsyncWebserver and AsyncTCP,  they include both heap corruptions, illegal instructions and LoadStore errors seemingly at random, however after a recent upgrade of the ESP32 core in the Arduino IDE, which upgraded the SDK from V3.1 to V3.2 these exceptions seem to have completely cleared up. 

Can anyone from Espressif please comment on the changes made between V3.1 and V3.2 of the SDK and if these were intended to improve the stability of the Async services. Has anyone else noticed this change in behaviour.

thehellmaker, 2019-05-05:
> We have a fairly complex application that uses both cores, one for WiFi and one for our application and we use SPIFFS to hold the web pages, js and css files that are served to the client browser. Up until a few days ago we had a very unstable operation of the application with regular exception errors when accessing the web pages via the browser. We are using jquery to provide internal web page upgrades on a timer in the page code so there is a regular communication (once every two to five seconds) between browser and the ESP32 server.
> 
> All the exceptions point into ESPAsyncWebserver and AsyncTCP, they include both heap corruptions, illegal instructions and LoadStore errors seemingly at random, however after a recent upgrade of the ESP32 core in the Arduino IDE, which upgraded the SDK from V3.2 to V3.3 these exceptions seem to have completely cleared up.
> 
> Can anyone from Espressif please comment on the changes made between V3.2 and V3.3 of the SDK and if these were intended to improve the stability of the Async services. Has anyone else noticed this change in behaviour.

I am having the same issue after upgrade from 3.2 to 3.3.

dvcomp, 2019-07-15:
I have the same error, but it depends on the browser type.
 Crome OK, Chrome on adndroid OK, Firefox OK, but IE or EDGE causes this:
18:14:42.581 -> E (75123745) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
18:14:42.581 -> E (75123745) task_wdt:  - async_tcp (CPU 0/1)
18:14:42.581 -> E (75123745) task_wdt: Tasks currently running:
18:14:42.581 -> E (75123745) task_wdt: CPU 0: IDLE0
18:14:42.581 -> E (75123745) task_wdt: CPU 1: loopTask
18:14:42.614 -> E (75123745) task_wdt: Aborting.
18:14:42.614 -> abort() was called at PC 0x400e4def on core 0

The file is 250k jpeg.
(Latest async tcp, latest asyn webserver).

WTF?
On wireshark the packet sending just stops in the middle at a random position.

closed 2019-10-05 by stale[bot]