me-no-dev/AsyncTCP #14: ESP32 Crash on TCP Close

psykokwak-com, 2018-05-14:
Hi all,
I use AsyncTCPWebserver and When I disconnect my websocket client after some time, my ESP32 crash with that backtrace : 

> CORRUPT HEAP: Bad head at 0x3ffceca8. Expected 0xabba1234 got 0x3ffcef30
> assertion "head != NULL" failed: file "C:/msys32/home/Jyce/esp/esp-idf/components/heap/multi_heap_poisoning.c", line 201, function: multi_heap_free
> abort() was called at PC 0x400d37cf on core 1
> 0x400d37cf: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
> 
> 
> Backtrace: 0x4008e014:0x3ffc59f0 0x4008e1ef:0x3ffc5a10 0x400d37cf:0x3ffc5a30 0x4008dd25:0x3ffc5a60 0x40082982:0x3ffc5a80 0x40082ecd:0x3ffc5aa0 0x4000bec7:0x3ffc5ac0 0x40137e0d:0x3ffc5ae0 0x40137ecf:0x3ffc5b00 0x401022a1:0x3ffc5b20 0x40131019:0x3ffc5b40
> 0x4008e014: invoke_abort at C:/msys32/home/Jyce/esp/esp-idf/components/esp32/panic.c:648
> 
> 0x4008e1ef: abort at C:/msys32/home/Jyce/esp/esp-idf/components/esp32/panic.c:648
> 
> 0x400d37cf: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
> 
> 0x4008dd25: multi_heap_free at C:/msys32/home/Jyce/esp/esp-idf/components/heap/multi_heap_poisoning.c:299
> 
> 0x40082982: heap_caps_free at C:/msys32/home/Jyce/esp/esp-idf/components/heap/heap_caps.c:123
> 
> 0x40082ecd: _free_r at C:/msys32/home/Jyce/esp/esp-idf/components/newlib/syscalls.c:42
> 
> 0x40137e0d: tcp_close_shutdown at C:/msys32/home/Jyce/esp/esp-idf/components/lwip/core/tcp.c:835
> 
> 0x40137ecf: tcp_close at C:/msys32/home/Jyce/esp/esp-idf/components/lwip/core/tcp.c:835
> 
> 0x401022a1: _tcp_close_api(tcpip_api_call*) at C:/msys32/home/Jyce/esp/arduino/components/arduino/libraries/AsyncTCP/src/AsyncTCP.cpp:785
> 
> 0x40131019: tcpip_thread at C:/msys32/home/Jyce/esp/esp-idf/components/lwip/api/tcpip.c:474

I use the head of master of espressif IDF, Arduino, AsyncWebserver and TCPAsync at the time I write this message.

me-no-dev, 2018-07-24:
can you try the latest code please? I just added something that might help

psykokwak-com, 2018-07-25:
Hi,
It looks better but I need to do some more test.
Thanks for your support.

psykokwak-com, 2018-09-25:
I still have this issue with the latest commit.

> CORRUPT HEAP: Bad head at 0x3f807cb0. Expected 0xabba1234 got 0x3f807dd0
> assertion "head != NULL" failed: file "C:/msys32/home/Jyce/esp/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: mu
> lti_heap_free
> abort() was called at PC 0x400d7187 on core 0
> 0x400d7187: __assert_func at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdli
> b/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
> 
> 
> Backtrace: 0x40096def:0x3ffcf370 0x40097040:0x3ffcf3a0 0x400d7187:0x3ffcf3d0 0x400967ed:0x3ffcf400 0x40085b71:0x3ffcf430 0x4008646e:0x
> 3ffcf460 0x4000bec7:0x3ffcf490 0x40160c61:0x3ffcf4b0 0x40160d60:0x3ffcf4e0 0x4011e92c:0x3ffcf510 0x40159226:0x3ffcf540
> 0x40096def: invoke_abort at C:/msys32/home/Jyce/esp/esp-idf/components/esp32/panic.c:664
> 
> 0x40097040: abort at C:/msys32/home/Jyce/esp/esp-idf/components/esp32/panic.c:664
> 
> 0x400d7187: __assert_func at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdli
> b/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
> 
> 0x400967ed: multi_heap_free at C:/msys32/home/Jyce/esp/esp-idf/components/heap/multi_heap_poisoning.c:301
> 
> 0x40085b71: heap_caps_free at C:/msys32/home/Jyce/esp/esp-idf/components/heap/heap_caps.c:351
> 
> 0x4008646e: _free_r at C:/msys32/home/Jyce/esp/esp-idf/components/newlib/syscalls.c:42
> 
> 0x40160c61: tcp_close_shutdown at C:/msys32/home/Jyce/esp/esp-idf/components/lwip/core/tcp.c:918
> 
> 0x40160d60: tcp_close at C:/msys32/home/Jyce/esp/esp-idf/components/lwip/core/tcp.c:918
> 
> 0x4011e92c: _tcp_close_api(tcpip_api_call*) at C:/msys32/home/Jyce/esp/arduino/components/arduino/libraries/AsyncTCP/src/AsyncTCP.cpp:
> 933
> 
> 0x40159226: tcpip_thread at C:/msys32/home/Jyce/esp/esp-idf/components/lwip/api/tcpip.c:481
> 
> 
> Rebooting...

bertmelis, 2018-12-14:
I think I'm suffering the same error:

```
CORRUPT HEAP: Bad head at 0x3ffbb530. Expected 0xabba1234 got 0x3ffbb600
assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x400d8d67 on core 0
```

I'm not using the webserver, only AsyncMqttClient. I just connect, send data en reconnect when the broker disconnects. The odd part is that the broker regularely disconnects because of a timeout while it shouldn't do that (my ESP just keeps sending data). Also: my ESP thinks it is still connected while the broker reports it as disconnected.

Somewhere down the line the ESP also throws this error:
```
[E][AsyncTCP.cpp:571] _recv(): 0x3ffc9e24 != 0x3ffbb9b8
```

I believe the broker disconnects a previous connection but the ESP thinks it is the current connection.

psykokwak-com, 2019-01-17:
Still have this issue with the latest master :/

zhivko, 2019-03-07:
I have similar problem. It seems it happens on AsyncTCP _tcp_close_api like this:

```
# make monitor
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/msys32/home/klemen/esp/esp-idf/requirements.txt are satisfied.
MONITOR
--- idf_monitor on COM28 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
--- Profiler version monitor
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5584
ho 0 tail 12 room 4
load:0x40078000,len:11028
load:0x40080400,len:6644
entry 0x40080760
I (30) boot: ESP-IDF v3.3-beta2-dirty 2nd stage bootloader
I (30)P(00) boot: compile time 16:47:34
I (41)P(11) boot: Enabling RNG early entropy source...
I (41)P(00) boot: SPI Speed      : 40MHz
I (41)P(00) boot: SPI Mode       : DIO
I (41)P(00) boot: SPI Flash Size : 4MB
I (44)P(03) boot: Partition Table:
I (47)P(03) boot: ## Label            Usage          Type ST Offset   Length
I (53)P(06) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (60)P(07) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (66)P(06) boot:  2 app0             OTA app          00 10 00010000 00140000
I (73)P(07) boot:  3 app1             OTA app          00 11 00150000 00140000
I (79)P(06) boot:  4 eeprom           Unknown data     01 99 00290000 00001000
I (86)P(07) boot:  5 spiffs           Unknown data     01 82 00291000 0015e000
I (92)P(06) boot:  6 coredump         Unknown data     01 03 003ef000 00010000
I (99)P(07) boot: End of partition table
I (102)P(03) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3c1a8 (246184) map
I (196)P(94) esp_image: segment 1: paddr=0x0004c1d0 vaddr=0x3ffb0000 size=0x03e40 ( 15936) load
I (203)P(07) esp_image: segment 2: paddr=0x00050018 vaddr=0x400d0018 size=0xaa77c (698236) map
0x400d0018: _flash_cache_start at ??:?

I (448)P!(245) esp_image: segment 3: paddr=0x000fa79c vaddr=0x3ffb3e40 size=0x0015c (   348) load
I (448)P(00) esp_image: segment 4: paddr=0x000fa900 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/msys32/home/klemen/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (453)P(05) esp_image: segment 5: paddr=0x000fad08 vaddr=0x40080400 size=0x19fd4 (106452) load
I (520)P(67) boot: Loaded app from partition at offset 0x10000
I (520)P(00) boot: Disabling RNG early entropy source...
I (520)P(00) cpu_start: Pro cpu up.
I (521)P(01) cpu_start: Application information:
I (525)P(04) cpu_start: Project name:     DoubleLifter
I (530)P(05) cpu_start: App version:      a7f6420-dirty
I (534)P(04) cpu_start: Compile time:     Mar  7 2019 10:21:03
I (539)P(05) cpu_start: ELF file SHA256:  3ea4115d84da6ea9...
I (544)P(05) cpu_start: ESP-IDF:          v3.3-beta2-dirty
I (549)P(05) cpu_start: Starting app cpu, entry point is 0x40081684
0x40081684: call_start_cpu1 at C:/msys32/home/klemen/esp/esp-idf/components/esp32/cpu_start.c:267

I (0) cpu_start: App cpu up.
I (558)P(09) heap_init: Initializing. RAM available for dynamic allocation:
I (563)P(05) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
[... 294 lines cut ...]
0x40082f86: heap_caps_free at C:/msys32/home/klemen/esp/esp-idf/components/heap/heap_caps.c:269

0x400842bd: _free_r at C:/msys32/home/klemen/esp/esp-idf/components/newlib/syscalls.c:42

0x401556b1: mem_free at C:/msys32/home/klemen/esp/esp-idf/components/lwip/lwip/src/core/mem.c:151

0x40155397: do_memp_free_pool at C:/msys32/home/klemen/esp/esp-idf/components/lwip/lwip/src/core/memp.c:459

0x401553e3: memp_free at C:/msys32/home/klemen/esp/esp-idf/components/lwip/lwip/src/core/memp.c:489

0x40154b32: tcp_close_shutdown at C:/msys32/home/klemen/esp/esp-idf/components/lwip/lwip/src/core/tcp.c:986

0x40154b9a: tcp_close at C:/msys32/home/klemen/esp/esp-idf/components/lwip/lwip/src/core/tcp.c:986

0x4017489d: _tcp_close_api(tcpip_api_call_data*) at C:/msys32/home/klemen/esp/DoubleLifter/components/AsyncTCP/src/AsyncTCP.cpp:876

0x4014b7b0: tcpip_thread at C:/msys32/home/klemen/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:483

0x40097f91: vPortTaskWrapper at C:/msys32/home/klemen/esp/esp-idf/components/freertos/port.c:435

I (1872)P(0-112) esp_core_dump_flash: Save core dump to flash...
I (1877)P(05) esp_core_dump_common: Found tasks: (12)!
I (2018)P!(141) esp_core_dump_flash: Core dump has been saved to flash.
Rebooting...
```

zhivko, 2019-03-07:
I also see that in some moment cpu_1 that runs async_tcp doesnt rest WDT. From ```make monitor``` I get this:
```
E (41724)P!(5000) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (41724)P(00) task_wdt:  - IDLE1 (CPU 1)
E (41724)P(00) task_wdt: Tasks currently running:
E (41724)P(00) task_wdt: CPU 0: main
E (41724)P(00) task_wdt: CPU 1: async_tcp
```
It seems like async_tcp blocks CPU 1.

closed 2019-10-05 by stale[bot]

mrxa13, 2021-01-13:
If think I have a similiar issue on closing a connection (in my case the connection of a mqtt client). 
Assigning all threads to a single core (most important the free RTOS core fixed other crashes on send before because lwip is not thread safe)

I did not find any hint, on how this issue on tcp_close was fixed. I think the issue still exists.

Here is my crash log:

[I][mqttlayer.cpp:74] onMqttDisconnect(): Disconnected from MQTT done
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c3f0  PS      : 0x00060930  A0      : 0x8013ba78  A1      : 0x3ffe1d40  
A2      : 0x00000000  A3      : 0x00000004  A4      : 0xfffffffc  A5      : 0x3f409ac5  
A6      : 0x3ffcf648  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000004  
A10     : 0x3ffce5a4  A11     : 0x3ffce574  A12     : 0x3ffce574  A13     : 0x00000001  
A14     : 0x0000cdcd  A15     : 0xffffff98  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

ELF file SHA256: 7bc66aa6a58aa347

Backtrace: 0x4000c3f0:0x3ffe1d40 0x4013ba75:0x3ffe1d60 0x4013bcf9:0x3ffe1d80 0x4013bd9f:0x3ffe1db0 0x401bbdd1:0x3ffe1de0 0x4013c6b3:0x3ffe1e00 0x4013c759:0x3ffe1e30 0x4013abe2:0x3ffe1e80 0x4013ac5a:0x3ffe1ee0 0x401bdc69:0x3ffe1f00 0x401b995b:0x3ffe1f30 0x401b8e7c:0x3ffe2360 0x401b8f49:0x3ffe23a0 0x401b93c2:0x3ffe23c0
0x4013ba75: std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>* std::__uninitialized_copy<false>::__uninit_copy<std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*, std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*>(std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*, std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*, std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*) at /home/mrxa/sketchbook/esp32/mka/components/async-mqtt-client/async-mqtt-client/src/AsyncMqttClient.cpp:106
 (inlined by) std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>* std::uninitialized_copy<std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*, std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*>(std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*, std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*, std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*) at /home/mrxa/sketchbook/esp32/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_uninitialized.h:126
 (inlined by) std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>* std::__uninitialized_copy_a<std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*, std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*, std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)> >(std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*, std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*, std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*, std::allocator<std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)> >&) at /home/mrxa/sketchbook/esp32/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_uninitialized.h:281
 (inlined by) std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>* std::__uninitialized_move_if_noexcept_a<std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*, std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*, std::allocator<std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)> > >(std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*, std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*, std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>*, std::allocator<std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)> >&) at /home/mrxa/sketchbook/esp32/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_uninitialized.h:304
 (inlined by) void std::vector<std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>, std::allocator<std::function<void (char*, char*, AsyncMqttClientMessageP [... cut ...]
0x401b995b: tcp_ssl_send at /home/mrxa/sketchbook/esp32/mka/components/AsyncTCP/src/tcp_mbedtls.c:520

0x401b8e7c: AsyncClient::_close() at /home/mrxa/sketchbook/esp32/mka/components/AsyncTCP/src/AsyncTCP.cpp:828

0x401b8f49: AsyncClient::add(char const*, unsigned int, unsigned char) at /home/mrxa/sketchbook/esp32/mka/components/AsyncTCP/src/AsyncTCP.cpp:828

0x401b93c2: std::function<void (void*, AsyncClient*, signed char)>::operator()(void*, AsyncClient*, signed char) const at /home/mrxa/sketchbook/esp32/mka/components/AsyncTCP/src/AsyncTCP.cpp:828

Rebooting...
ets Jun  8 2016 00:22:57