me-no-dev/AsyncTCP #19: Errors compiling after Arduino core update to e5ea089a7fd5cc8a96d68aae64dc3b4dcef2315e

CelliesProjects, 2018-09-21:
After updating Arduino Core to commit e5ea089a7fd5cc8a96d68aae64dc3b4dcef2315e AsyncTCP does not compile anymore.

A whole lot of errors:
```
Compiling library "AsyncTCP"
"/home/cellie/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/config" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/bluedroid/api" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/app_trace" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/app_update" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/bootloader_support" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/bt" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/driver" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/esp_adc_cal" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/esp_http_client" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/esp_https_ota" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/esp-mqtt" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/esp-tls" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/ethernet" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/fatfs" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/heap" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/http_server" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/jsmn" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/log" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/mdns" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls_port" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/nvs_flash" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/openssl" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/spi_flash" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/sdmmc" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/smartconfig_ack" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/spiffs" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/tcp_transport" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/ulp" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/vfs" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/wear_levelling" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/xtensa-debug-module" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/lwip" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/coap" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/console" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/expat" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/json" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/nghttp" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/soc" "-I/home/cellie/Arduino/hardware/espressif/esp32/tools/sdk/include/wpa_supplicant" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib  -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti -MMD -c -DF_CPU=240000000L -DARDUINO=10805 -DARDUINO_MH_ET_LIVE_ESP32MINIKIT -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="MH_ET_LIVE_ESP32MINIKIT" -DARDUINO_VARIANT="mhetesp32minikit"  -DESP32 -DCORE_DEBUG_LEVEL=3  "-I/home/cellie/Arduino/hardware/espressif/esp32/cores/esp32" "-I/home/cellie/Arduino/hardware/espressif/esp32/variants/mhetesp32minikit" "-I/home/cellie/Arduino/hardware/espressif/esp32/libraries/SPI/src" "-I/home/cellie/Arduino/hardware/espressif/esp32/libraries/SPIFFS/src" "-I/home/cellie/Arduino/hardware/espressif/esp32/libraries/FS/src" "-I/home/cellie/Arduino/hardware/espressif/esp32/libraries/ESPmDNS/src" "-I/home/cellie/Arduino/hardware/espressif/esp32/libraries/Preferences/src" "-I/home/cellie/Arduino/libraries/Adafruit_ILI9341" "-I/home/cellie/Arduino/libraries/Adafruit_GFX_Library" "-I/home/cellie/Arduino/libraries/OneWire" "-I/home/cellie/Arduino/libraries/ESP8266_and_ESP32_Oled_Driver_for_SSD1306_display/src" "-I/home/cellie/Arduino/hardware/espressif/esp32/libraries/Wire/src" "-I/home/cellie/Arduino/libraries/AsyncTCP/src" "-I/home/cellie/Arduino/libraries/ESPAsyncWebServer/src" "-I/home/cellie/Arduino/hardware/espressif/esp32/libraries/WiFi/src" "-I/home/cellie/Arduino/libraries/XPT2046_Touchscreen" "-I/home/cellie/Arduino/libraries/MoonPhase" "/home/cellie/Arduino/libraries/AsyncTCP/src/AsyncTCP.cpp" -o "/tmp/arduino_build_820309/libraries/AsyncTCP/AsyncTCP.cpp.o"
/home/cellie/Arduino/libraries/AsyncTCP/src/AsyncTCP.cpp:195:27: error: field 'call' has incomplete type 'tcpip_api_call'
     struct tcpip_api_call call;
                           ^
/home/cellie/Arduino/libraries/AsyncTCP/src/AsyncTCP.cpp:195:12: note: forward declaration of 'struct tcpip_api_call'
     struct tcpip_api_call call;
            ^
/home/cellie/Arduino/libraries/AsyncTCP/src/AsyncTCP.cpp: In function 'esp_err_t _tcp_output(tcp_pcb*)':
/home/cellie/Arduino/libraries/AsyncTCP/src/AsyncTCP.cpp:227:65: error: invalid conversion from 'err_t (*)(tcpip_api_call*) {aka signed char (*)(tcpip_api_call*)}' to 'tcpip_api_call_fn {aka signed char (*)(tcpip_api_call_data*)}' [-fpermissive]
     tcpip_api_call(_tcp_output_api, (struct tcpip_api_call*)&msg);
                                                                 ^
/home/cellie/Arduino/libraries/AsyncTCP/src/AsyncTCP.cpp:227:65: error: cannot convert 'tcpip_api_call*' to 'tcpip_api_call_data*' for argument '2' to 'err_t tcpip_api_call(tcpip_api_call_fn, tcpip_api_call_data*)'
/home/cellie/Arduino/libraries/AsyncTCP/src/AsyncTCP.cpp:195:12: note: class type 'tcpip_api_call' is incomplete
     struct tcpip_api_call call;
            ^
/home/cellie/Arduino/libraries/AsyncTCP/src/AsyncTCP.cpp: In function 'esp_err_t _tcp_write(tcp_pcb*, const char*, size_t, uint8_t)':
/home/cellie/Arduino/libraries/AsyncTCP/src/AsyncTCP.cpp:243:64: error: invalid conversion from 'err_t (*)(tcpip_api_call*) {aka signed char (*)(tcpip_api_call*)}' to 'tcpip_api_call_fn {aka signed char (*)(tcpip_api_call_data*)}' [-fpermissive]
     tcpip_api_call(_tcp_write_api, (struct tcpip_api_call*)&msg);
                                                                ^
/home/cellie/Arduino/libraries/AsyncTCP/src/AsyncTCP.cpp:243:64: error: cannot convert 'tcpip_api_call*' to 'tcpip_api_call_data*' for argument '2' to 'err_t tcpip_api_call(tcpip_api_call_fn, tcpip_api_call_data*)'
/home/cellie/Arduino/libraries/AsyncTCP/src/AsyncTCP.cpp:195:12: note: class type 'tcpip_api_call' is incomplete
     struct tcpip_api_call call;
            ^
/home/cellie/Arduino/libraries/AsyncTCP/src/AsyncTCP.cpp: In function 'esp_err_t _tcp_recved(tcp_pcb*, size_t)':
/home/cellie/Arduino/libraries/AsyncTCP/src/AsyncTCP.cpp:258:65: error: invalid conversion from 'err_t (*)(tcpip_api_call*) {aka signed char (*)(tcpip_api_call*)}' to 'tcpip_api_call_fn {aka signed char (*)(tcpip_api_call_data*)}' [-fpermissive]
     tcpip_api_call(_tcp_recved_api, (struct tcpip_api_call*)&msg);
                                                                 ^
/home/cellie/Arduino/libraries/AsyncTCP/src/AsyncTCP.cpp:258:65: error: cannot convert 'tcpip_api_call*' to 'tcpip_api_call_data*' for argument '2' to 'err_t tcpip_api_call(tcpip_api_call_fn, tcpip_api_call_data*)'
[... cut ...]
/home/cellie/Arduino/libraries/AsyncTCP/src/AsyncTCP.cpp:873:6: error: prototype for 'void AsyncClient::_s_dns_found(const char*, ip_addr_t*, void*)' does not match any in class 'AsyncClient'
 void AsyncClient::_s_dns_found(const char * name, ip_addr_t * ipaddr, void * arg){
      ^
In file included from /home/cellie/Arduino/libraries/AsyncTCP/src/AsyncTCP.cpp:24:0:
/home/cellie/Arduino/libraries/AsyncTCP/src/AsyncTCP.h:164:17: error: candidate is: static void AsyncClient::_s_dns_found(const char*, _ip_addr*, void*)
     static void _s_dns_found(const char *name, struct _ip_addr *ipaddr, void *arg);
```
Could be an Arduino Core problem, if so I will open a issue there.

luc-github, 2018-09-21:
update with https://github.com/me-no-dev/AsyncTCP/tree/idf-update

CelliesProjects, 2018-09-22:
Thanks, that fixed it.

closed 2018-09-22 by CelliesProjects

eziosoma, 2019-01-12:
What are di command thet I have to execute for updating PlatformIO to https://github.com/me-no-dev/AsyncTCP/tree/idf-update?

elfege, 2019-02-17:
> update with https://github.com/me-no-dev/AsyncTCP/tree/idf-update

this link is dead... I'm stuck, how did you get it to work? What is platformio.ini? I have no such file anywhere... let me know if you can help...

atanisoft, 2019-02-17:
> > update with https://github.com/me-no-dev/AsyncTCP/tree/idf-update
> 
> this link is dead... I'm stuck, how did you get it to work? What is platformio.ini? I have no such file anywhere... let me know if you can help...

Of course the link is dead, the branch was deleted after it was merged (and that link was posted in Sept 2018!) platformio.ini is used by PlatformIO and not Arduino IDE so you can ignore that.

In Arduino IDE you need to make sure you are using the latest arduino-esp32 (1.0.1) and AsyncTCP (1.0.3). If you are still having issues please open a new issue rather than comment on a very old one.

elfege, 2019-02-17:
> @elfege
> 
> > > update with https://github.com/me-no-dev/AsyncTCP/tree/idf-update
> > 
> > 
> > this link is dead... I'm stuck, how did you get it to work? What is platformio.ini? I have no such file anywhere... let me know if you can help...
> 
> Of course the link is dead, the branch was deleted after it was merged (and that link was posted in Sept 2018!) platformio.ini is used by PlatformIO and not Arduino IDE so you can ignore that.
> 
> In Arduino IDE you need to make sure you are using the latest arduino-esp32 (1.0.1) and AsyncTCP (1.0.3). If you are still having issues please open a new issue rather than comment on a very old one.

Thanks for the clarification. I've read about the merge but it wasn't clear to me since I have reinstalled the esp32 environment and got the latest AsyncTCP version and still not compiling... Ok to open a new thread but I'm pretty sure I'll be rerouted to this one as this is what usually happens. Thanks a lot for you answer!

atanisoft, 2019-02-17:
@elfege
> Thanks for the clarification. I've read about the merge but it wasn't clear to me since I have reinstalled the esp32 environment and got the latest AsyncTCP version and still not compiling... Ok to [...]

You may just need to remove the library from the Arduino IDE library manager and re-add it. Most likely the IDE is using a cached copy of the library and not pulling the latest pieces.

elfege, 2019-02-17:
Just did that... no joy... It's getting weird (thanks so much for your help!!!)
[image]

atanisoft, 2019-02-17:
I'd say open a new issue for tracking and in that we can sort through your issue there. Please be sure to include full compilation output and versions for arduino-esp32 and AsyncTCP.

elfege, 2019-02-17:
I'll wait until I've verified I haven't messed something up somewhere when updated to ESP1.0.1. Judging by the fact this is an old issue it's much more likely a problem on my end.

elfege, 2019-02-17:
It's working now. I don't know what went wrong bu even after recompiling the core it wouldn't work so I reinstalled it entirely and now it works. Thanks so much @atanisoft for showing the right direction!