me-no-dev/AsyncTCP #21: Error during compilation with newest arduino-esp32

smacyas, 2018-09-22:
WROOM -32
Arduino IDE

Until the last update espressif/arduino-esp32, there were no errors during compilation, and now the project is not compiled.

C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:259:27: error: field 'call' has incomplete type 'tcpip_api_call'
     struct tcpip_api_call call;
                          ^
C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:259:12: note: forward declaration of 'struct tcpip_api_call'
     struct tcpip_api_call call;

            ^
C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp: In function 'esp_err_t _tcp_output(tcp_pcb*)':

C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:291:65: error: invalid conversion from 'err_t (*)(tcpip_api_call*) {aka signed char (*)(tcpip_api_call*)}' to 'tcpip_api_call_fn {aka signed char (*)(tcpip_api_call_data*)}' [-fpermissive]

     tcpip_api_call(_tcp_output_api, (struct tcpip_api_call*)&msg);

                                                            ^

C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:291:65: error: cannot convert 'tcpip_api_call*' to 'tcpip_api_call_data*' for argument '2' to 'err_t tcpip_api_call(tcpip_api_call_fn, tcpip_api_call_data*)'

C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:259:12: note: class type 'tcpip_api_call' is incomplete

     struct tcpip_api_call call;

            ^

C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp: In function 'esp_err_t _tcp_write(tcp_pcb*, const char*, size_t, uint8_t)':

C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:307:64: error: invalid conversion from 'err_t (*)(tcpip_api_call*) {aka signed char (*)(tcpip_api_call*)}' to 'tcpip_api_call_fn {aka signed char (*)(tcpip_api_call_data*)}' [-fpermissive]

     tcpip_api_call(_tcp_write_api, (struct tcpip_api_call*)&msg);

                                                                ^

C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:307:64: error: cannot convert 'tcpip_api_call*' to 'tcpip_api_call_data*' for argument '2' to 'err_t tcpip_api_call(tcpip_api_call_fn, tcpip_api_call_data*)'

C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:259:12: note: class type 'tcpip_api_call' is incomplete

     struct tcpip_api_call call;

            ^

C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp: In function 'esp_err_t _tcp_recved(tcp_pcb*, size_t)':

C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:322:65: error: invalid conversion from 'err_t (*)(tcpip_api_call*) {aka signed char (*)(tcpip_api_call*)}' to 'tcpip_api_call_fn {aka signed char (*)(tcpip_api_call_data*)}' [-fpermissive]

     tcpip_api_call(_tcp_recved_api, (struct tcpip_api_call*)&msg);

                                                                 ^

C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:322:65: error: cannot convert 'tcpip_api_call*' to 'tcpip_api_call_data*' for argument '2' to 'err_t tcpip_api_call(tcpip_api_call_fn, tcpip_api_call_data*)'

C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:259:12: note: class type 'tcpip_api_call' is incomplete

     struct tcpip_api_call call;

            ^

C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp: In function 'esp_err_t _tcp_connect(tcp_pcb*, ip_addr_t*, uint16_t, tcp_connected_fn)':

C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:338:66: error: invalid conversion from 'err_t (*)(tcpip_api_call*) {aka signed char (*)(tcpip_api_call*)}' to 'tcpip_api_call_fn {aka signed char (*)(tcpip_api_call_data*)}' [-fpermissive]

     tcpip_api_call(_tcp_connect_api, (struct tcpip_api_call*)&msg);

                                                                  ^

C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:338:66: error: cannot convert 'tcpip_api_call*' to 'tcpip_api_call_data*' for argument '2' to 'err_t tcpip_api_call(tcpip_api_call_fn, tcpip_api_call_data*)'

[... 77 lines cut ...]

In file included from C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:24:0:

C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.h:84:10: error: candidate is: void AsyncClient::_dns_found(_ip_addr*)

     void _dns_found(struct _ip_addr *ipaddr);

          ^

C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:946:6: error: prototype for 'void AsyncClient::_s_dns_found(const char*, ip_addr_t*, void*)' does not match any in class 'AsyncClient'

 void AsyncClient::_s_dns_found(const char * name, ip_addr_t * ipaddr, void * arg){

      ^

In file included from C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:24:0:

C:\Users\Matsys\Documents\Arduino\hardware\espressif\esp32\libraries\AsyncTCP\src\AsyncTCP.h:164:17: error: candidate is: static void AsyncClient::_s_dns_found(const char*, _ip_addr*, void*)

     static void _s_dns_found(const char *name, struct _ip_addr *ipaddr, void *arg);

                 ^

exit status 1
Ошибка компиляции для платы ESP32 Dev Module.

closed 2018-09-24 by smacyas

elfege, 2019-02-17:
Thank you for sending us to a different page where they already talk about this issue, except it is for PlatformIO IDE and here we're stuck with an Arduino IDE problem... How do we get this fixed in Arduino ? Any idea ?

AMC-Electronics, 2019-02-23:
I'm having the same issue when trying to use AsyncTCP with Arduino IDE.

elfege, 2019-02-24:
I figured it out. Install / delete your old core and install the latest version. It is now directly at bailable through IDE board management. No need to go through the whole script/shell again.

ctshh, 2019-03-19:
> I figured it out. Install / delete your old core and install the latest version. It is now directly at bailable through IDE board management. No need to go through the whole script/shell again.

+1 here. Thanks!

Wzdforever, 2019-12-23:
this question is still bother me

Wzdforever, 2019-12-23:
Arduino:1.8.10 (Windows 10), 开发板："ESP-32F, 80MHz, 921600"

N:\arduino\hardware\arduino-esp32\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:84:37: error: 'CONFIG_LWIP_MAX_ACTIVE_TCP' was not declared in this scope

 const int _number_of_closed_slots = CONFIG_LWIP_MAX_ACTIVE_TCP;

                                     ^

N:\arduino\hardware\arduino-esp32\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:85:54: error: size of array '_closed_slots' is not an integral constant-expression

 static uint32_t _closed_slots[_number_of_closed_slots];

                                                      ^

N:\arduino\hardware\arduino-esp32\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp: In function 'void _handle_async_event(lwip_event_packet_t*)':

N:\arduino\hardware\arduino-esp32\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:183:68: error: no matching function for call to 'AsyncClient::_s_dns_found(const char*&, ip_addr_t*, void*&)'

         AsyncClient::_s_dns_found(e->dns.name, &e->dns.addr, e->arg);

                                                                    ^

In file included from N:\arduino\hardware\arduino-esp32\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:24:0:

N:\arduino\hardware\arduino-esp32\esp32\libraries\AsyncTCP\src\AsyncTCP.h:137:17: note: candidate: static void AsyncClient::_s_dns_found(const char*, ip_addr*, void*)

     static void _s_dns_found(const char *name, struct ip_addr *ipaddr, void *arg);

                 ^

N:\arduino\hardware\arduino-esp32\esp32\libraries\AsyncTCP\src\AsyncTCP.h:137:17: note:   no known conversion for argument 2 from 'ip_addr_t* {aka _ip_addr*}' to 'ip_addr*'

N:\arduino\hardware\arduino-esp32\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp: In function 'void _async_service_task(void*)':

N:\arduino\hardware\arduino-esp32\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:193:37: error: 'esp_task_wdt_add' was not declared in this scope

             if(esp_task_wdt_add(NULL) != ESP_OK){

                                     ^

N:\arduino\hardware\arduino-esp32\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:199:40: error: too many arguments to function 'void esp_task_wdt_delete()'

             if(esp_task_wdt_delete(NULL) != ESP_OK){

                                        ^

In file included from N:\arduino\hardware\arduino-esp32\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:32:0:

N:\arduino\hardware\arduino-esp32\esp32/tools/sdk/include/esp32/esp_task_wdt.h:61:6: note: declared here

 void esp_task_wdt_delete();

      ^

In file included from N:\arduino\hardware\arduino-esp32\esp32/tools/sdk/include/esp32/esp_system.h:20:0,

                 from N:\arduino\hardware\arduino-esp32\esp32\cores\esp32/esp32-hal.h:62,

                 from N:\arduino\hardware\arduino-esp32\esp32\cores\esp32/Arduino.h:35,

                 from N:\arduino\hardware\arduino-esp32\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:22:

N:\arduino\hardware\arduino-esp32\esp32/tools/sdk/include/esp32/esp_err.h:28:25: error: invalid operands of types 'void' and 'int' to binary 'operator!='

 #define ESP_OK          0

                         ^

N:\arduino\hardware\arduino-esp32\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:199:45: note: in expansion of macro 'ESP_OK'

[... 164 lines cut ...]
N:\arduino\hardware\arduino-esp32\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:981:24: error: invalid use of incomplete type 'struct ip_addr'

     if(ipaddr && ipaddr->u_addr.ip4.addr){

                        ^

In file included from N:\arduino\hardware\arduino-esp32\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:24:0:

N:\arduino\hardware\arduino-esp32\esp32\libraries\AsyncTCP\src\AsyncTCP.h:53:8: note: forward declaration of 'struct ip_addr'

 struct ip_addr;

        ^

N:\arduino\hardware\arduino-esp32\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:982:33: error: invalid use of incomplete type 'struct ip_addr'

         connect(IPAddress(ipaddr->u_addr.ip4.addr), _connect_port);

                                 ^

In file included from N:\arduino\hardware\arduino-esp32\esp32\libraries\AsyncTCP\src\AsyncTCP.cpp:24:0:

N:\arduino\hardware\arduino-esp32\esp32\libraries\AsyncTCP\src\AsyncTCP.h:53:8: note: forward declaration of 'struct ip_addr'

 struct ip_addr;