me-no-dev/AsyncTCP #28: compilation error on linux

erfan1383, 2018-11-29:


d0ct0rd-dalimo, 2018-12-13:
Hi!

I have the same issue after ESP32 Arduino core and libs updating. You should downgrade WiFi lib for ESP32. It's so sad, because I want to connect to enterprise net, it works on latest version of ESP32 lib, AsyncTCP not:(

ccoenen, 2018-12-22:
Same here on windows 10 with Arduino IDE `1.8.8` and esp32-arduino `1.0.1-rc2`

ccoenen, 2018-12-22:
I can also confirm that the suggestion proposed in #24 compiles fine: Use the [idf-update branch](https://github.com/me-no-dev/AsyncTCP/tree/idf-update).

I haven't tested it beyond compiling, though!

abeco, 2019-04-09:
I must be missing something, I'm getting the same issue but I don't see any idf-update branch. the link @ccoenen posted leads nowhere

ccoenen, 2019-04-10:
That branch has been merged into master: https://github.com/me-no-dev/AsyncTCP/commit/1656fed6fbd8bef515d3ad6caf1bf9d96ee3fe06 so the master should compile.

closed 2019-10-05 by stale[bot]