me-no-dev/AsyncTCP #39: Thread safety/multicore and race conditions

BlackEdder, 2019-04-15:
My debugging seems to indicate that AsyncTCP is not thread/multicore save on the ESP32. I.e. one or more of the onError/onDisconnect/onData etc events will run parallel with the main core on the ESP32 and can therefore cause race conditions. I solved this by wrapping all events in a mutex and since then the code has been stable. More [background](https://gitlab.com/painlessMesh/painlessMesh/issues/232).

I was wondering if this was by design? If so it might be good to mention it in the README/code comments.

I think this might be the cause of: https://github.com/me-no-dev/ESPAsyncWebServer/issues/324

trycoon, 2019-04-21:
Do you have a patch so the rest of us could try it out, I also suffer from occasional crashes.
Better yet, provide a pull-request so this could be merged.

BlackEdder, 2019-04-23:
I am afraid I fixed it by wrapping the interactions between my own code and the AsyncTCP in a mutex on my side of the code (which is extremely usage specific). Not familiar enough with the AsyncTCP code to attempt it inside the library (although from a quick scan: replacing the _in_lwip_thread checks with actual mutexes might do it).

me-no-dev, 2019-06-22:
i just pushed an update. I hope it is OK now.

closed 2019-10-05 by stale[bot]