me-no-dev/AsyncTCP #44: Code does not compile

lesept777, 2019-05-01:
I want to try a code found here : https://techtutorialsx.com/

Here is the error message
`Arduino : 1.8.9 (Windows 10), Carte : "LOLIN D32, Default, 80MHz, 921600, None"

C:\Users\Chuwi\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp:259:32: error: field 'call' has incomplete type 'tcpip_api_call_data'

     struct tcpip_api_call_data call;

                                ^

C:\Users\Chuwi\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp:259:12: note: forward declaration of 'struct tcpip_api_call_data'

     struct tcpip_api_call_data call;

            ^

C:\Users\Chuwi\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp: In function 'esp_err_t _tcp_output(tcp_pcb*)':

C:\Users\Chuwi\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp:295:70: error: invalid conversion from 'err_t (*)(tcpip_api_call_data*) {aka signed char (*)(tcpip_api_call_data*)}' to 'tcpip_api_call_fn {aka signed char (*)(tcpip_api_call*)}' [-fpermissive]

     tcpip_api_call(_tcp_output_api, (struct tcpip_api_call_data*)&msg);

                                                                      ^

C:\Users\Chuwi\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp:295:70: error: cannot convert 'tcpip_api_call_data*' to 'tcpip_api_call*' for argument '2' to 'err_t tcpip_api_call(tcpip_api_call_fn, tcpip_api_call*)'

C:\Users\Chuwi\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp:259:12: note: class type 'tcpip_api_call_data' is incomplete

     struct tcpip_api_call_data call;

            ^

C:\Users\Chuwi\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp: In function 'esp_err_t _tcp_write(tcp_pcb*, const char*, size_t, uint8_t)':

C:\Users\Chuwi\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp:311:69: error: invalid conversion from 'err_t (*)(tcpip_api_call_data*) {aka signed char (*)(tcpip_api_call_data*)}' to 'tcpip_api_call_fn {aka signed char (*)(tcpip_api_call*)}' [-fpermissive]

     tcpip_api_call(_tcp_write_api, (struct tcpip_api_call_data*)&msg);

                                                                     ^

C:\Users\Chuwi\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp:311:69: error: cannot convert 'tcpip_api_call_data*' to 'tcpip_api_call*' for argument '2' to 'err_t tcpip_api_call(tcpip_api_call_fn, tcpip_api_call*)'

C:\Users\Chuwi\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp:259:12: note: class type 'tcpip_api_call_data' is incomplete

     struct tcpip_api_call_data call;

            ^

C:\Users\Chuwi\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp: In function 'esp_err_t _tcp_recved(tcp_pcb*, size_t)':

C:\Users\Chuwi\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp:326:70: error: invalid conversion from 'err_t (*)(tcpip_api_call_data*) {aka signed char (*)(tcpip_api_call_data*)}' to 'tcpip_api_call_fn {aka signed char (*)(tcpip_api_call*)}' [-fpermissive]

     tcpip_api_call(_tcp_recved_api, (struct tcpip_api_call_data*)&msg);

                                                                      ^

C:\Users\Chuwi\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp:326:70: error: cannot convert 'tcpip_api_call_data*' to 'tcpip_api_call*' for argument '2' to 'err_t tcpip_api_call(tcpip_api_call_fn, tcpip_api_call*)'

C:\Users\Chuwi\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp:259:12: note: class type 'tcpip_api_call_data' is incomplete

     struct tcpip_api_call_data call;

            ^

C:\Users\Chuwi\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp: In function 'esp_err_t _tcp_connect(tcp_pcb*, ip_addr_t*, uint16_t, tcp_connected_fn)':

C:\Users\Chuwi\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp:342:71: error: invalid conversion from 'err_t (*)(tcpip_api_call_data*) {aka signed char (*)(tcpip_api_call_data*)}' to 'tcpip_api_call_fn {aka signed char (*)(tcpip_api_call*)}' [-fpermissive]

     tcpip_api_call(_tcp_connect_api, (struct tcpip_api_call_data*)&msg);
[... 124 lines cut ...]

  server.on("/hello", HTTP_GET, [](AsyncWebServerRequest * request) {
    request->send(200, "text/plain", "Hello World");
  });

  server.on("/printIp", HTTP_GET, [](AsyncWebServerRequest * request) {
    request->send(200, "text/plain", "ok");
    Serial.print("Received request from client with IP: ");
    Serial.println(request->client()->remoteIP());
  });

  server.on(
    "/post", HTTP_POST, [](AsyncWebServerRequest * request) {}, NULL,
  [](AsyncWebServerRequest * request, uint8_t *data, size_t len, size_t index, size_t total) {
    for (size_t i = 0; i < len; i++) Serial.write(data[i]);
    Serial.println();
    request->send(200);
  });

  server.begin();
}

void loop() {}

I'm using the latest IDE version 1.8.9 (but I had the same issue with v1.8.0) and selected either LOLIN D32 or ESP32 dev module

gtalusan, 2019-05-25:
I'm having the same problem using the ESP32 1.0.3-rc1 Arduino package.  Reverting 6fab7724abe9fb1669a77e8a346d12ec148ff19a in this repository makes it compile.

atanisoft, 2019-06-25:
If you are still encountering this with arduino-esp32 1.0.2 or 1.0.3-rc1 you are not compiling against that version. You will need to clear the arduino IDE cached libraries and platform versions and ensure it is compiling against the latest versions.

lesept777, 2019-06-25:
I went back to 1.8.5 and everything is fine now, thanks

atanisoft, 2019-06-25:
@lesept777 if your issue is fixed please consider closing this issue.

closed 2019-06-25 by lesept777