me-no-dev/AsyncTCP #47: LwIP/TCP thread uses 16KB stack

tve, 2019-06-25:
The stack size of the LwIP/TCP thread got increased to 16KB, which is quite a memory hog.
Is there a bug that should be fixed to reduce stack size, or perhaps something should be allocated dynamically so it can fail without crashing the whole system?
See https://github.com/me-no-dev/AsyncTCP/blob/master/src/AsyncTCP.cpp#L201
Introduced in commit https://github.com/me-no-dev/AsyncTCP/commit/f1a801fe069eda040d635aa877baf9efe093544c

tve, 2019-09-21:
Can you comment what drives the size of this stack? Where is what allocated that needs so much space?

me-no-dev, 2019-09-22:
i did it because we had many memory issues. can go down now again. want to make a PR?

danielkucera, 2019-11-10:
I'm hitting similar problem with esphome with wifi and BT stack enabled: there seems to be not enough memory to initialize AsyncTCP:
```
[01:35:24][E][AsyncTCP.cpp:1271] begin(): failed to start task
```

tve, 2020-01-09:
I guess I need to make a PR...

closed 2020-03-23 by stale[bot]

pbolduc, 2021-08-29:
> The stack size of the LwIP/TCP thread got increased to 16KB, which is quite a memory hog.
> Is there a bug that should be fixed to reduce stack size, or perhaps something should be allocated dynamically so it can fail without crashing the whole system?
> See https://github.com/me-no-dev/AsyncTCP/blob/master/src/AsyncTCP.cpp#L201
> Introduced in commit [f1a801f](https://github.com/me-no-dev/AsyncTCP/commit/f1a801fe069eda040d635aa877baf9efe093544c)

The stack size parameter in creating the task is in words not bytes. So (8192 * 2) words = (8192 * 2 * 4) = 64KB.  [xTaskCreate](https://www.freertos.org/a00125.html) 
> 
<html>
<body>

usStackDepth | The number of words (not bytes!) to allocate for use as the task's stack.
-- | --

</body>
</html>