me-no-dev/AsyncTCP #49: error: 'xTaskCreateUniversal' was not declared in this scope

jptrsn, 2019-06-27:
Trying to use the library for ESP32 OTA support. Newest version throws an exception that did not exist in version 1.0.3.

```
.piolibdeps\AsyncTCP_ID1826\src\AsyncTCP.cpp: In function 'bool _start_async_task()':
.piolibdeps\AsyncTCP_ID1826\src\AsyncTCP.cpp:205:141: error: 'xTaskCreateUniversal' was not declared in this scope
```

atanisoft, 2019-06-27:
This is due to PlatformIO not picking up arduino 1.0.2 which introduced this API. It is currently using arduino-esp32 1.0.1 as of framework-espressif32 1.8.0.

You can workaround this by changing your platformio.ini as follows:
```
#platform=espressif32
platform=https://github.com/platformio/platform-espressif32.git#feature/stage
```

pfeerick, 2019-06-30:
Incorrect. It is currently using arduino-esp32 1.0.2 as of platform-espressif32 1.8.0, which was released in April, and 1.9.0 was released two days ago, and still uses arduino-esp32 1.0.2.  The OP merely needs to ensure he runs `platformio update` in order to ensure he has current platform files, or in his projects `platformio.ini` change `platform = espressif32` to `platform = espressif32@>1.8.0` and platformio will ensure the right platform version is used.

atanisoft, 2019-06-30:
@pfeerick I just did a fresh install of 1.9.0 and it failed there as well. I've reached out the pio maintainer about this after that version came out.

pfeerick, 2019-06-30:
Can you provide a Minimal Complete Reproducible Example which demonstrates this?

Below is the output from my system... As you can see, it's using 1.9.0 of platformio-esp32 and 1.1.0 of AsyncTCP. This is when compiling the [simple_server example from ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/examples/simple_server/simple_server.ino) with no changes, and is successful. 1.8.0 is also, but 1.7.0 will fail with the `xTaskCreateUniversal` error. I'm using the platformio 4.0 beta for this log, as it gives the versions of platforms and libraries, but the stable build (3.6.7 IIRC) is also working fine also. 

platformio.ini
```
[env:mhetesp32minikit]
platform = espressif32@1.9.0 
board = mhetesp32minikit
framework = arduino
lib_deps = 
  ESP Async WebServer
```

Compile log
```
Processing mhetesp32minikit (platform: espressif32@1.9.0; board: mhetesp32minikit; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/mhetesp32minikit.html
PLATFORM: Espressif 32 1.9.0 > MH ET LIVE ESP32MiniKit
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10002.190628 (1.0.2), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 28 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP Async WebServer> 1.2.2
|   |-- <AsyncTCP> 1.1.0
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <AsyncTCP> 1.1.0
|-- <WiFi> 1.0
Compiling .pio/build/mhetesp32minikit/src/main.cpp.o
Generating partitions .pio/build/mhetesp32minikit/partitions.bin
Compiling .pio/build/mhetesp32minikit/lib515/AsyncTCP_ID1826/AsyncTCP.cpp.o
Compiling .pio/build/mhetesp32minikit/lib114/WiFi/ETH.cpp.o
Compiling .pio/build/mhetesp32minikit/lib114/WiFi/WiFi.cpp.o
Compiling .pio/build/mhetesp32minikit/lib114/WiFi/WiFiAP.cpp.o
Compiling .pio/build/mhetesp32minikit/lib114/WiFi/WiFiClient.cpp.o
Compiling .pio/build/mhetesp32minikit/lib114/WiFi/WiFiGeneric.cpp.o
Archiving .pio/build/mhetesp32minikit/lib515/libAsyncTCP_ID1826.a
Indexing .pio/build/mhetesp32minikit/lib515/libAsyncTCP_ID1826.a
Compiling .pio/build/mhetesp32minikit/lib114/WiFi/WiFiMulti.cpp.o
Compiling .pio/build/mhetesp32minikit/lib114/WiFi/WiFiSTA.cpp.o
Compiling .pio/build/mhetesp32minikit/lib114/WiFi/WiFiScan.cpp.o
Compiling .pio/build/mhetesp32minikit/lib114/WiFi/WiFiServer.cpp.o
Compiling .pio/build/mhetesp32minikit/lib114/WiFi/WiFiUdp.cpp.o
Compiling .pio/build/mhetesp32minikit/lib447/FS/FS.cpp.o
Compiling .pio/build/mhetesp32minikit/lib447/FS/vfs_api.cpp.o
Compiling .pio/build/mhetesp32minikit/lib3e6/ESP Async WebServer_ID306/AsyncEventSource.cpp.o
Archiving .pio/build/mhetesp32minikit/lib114/libWiFi.a
Indexing .pio/build/mhetesp32minikit/lib114/libWiFi.a
Compiling .pio/build/mhetesp32minikit/lib3e6/ESP Async WebServer_ID306/AsyncWebSocket.cpp.o
Compiling .pio/build/mhetesp32minikit/lib3e6/ESP Async WebServer_ID306/SPIFFSEditor.cpp.o
Archiving .pio/build/mhetesp32minikit/lib447/libFS.a
Indexing .pio/build/mhetesp32minikit/lib447/libFS.a
Compiling .pio/build/mhetesp32minikit/lib3e6/ESP Async WebServer_ID306/WebAuthentication.cpp.o
Compiling .pio/build/mhetesp32minikit/lib3e6/ESP Async WebServer_ID306/WebHandlers.cpp.o
Compiling .pio/build/mhetesp32minikit/lib3e6/ESP Async WebServer_ID306/WebRequest.cpp.o
Compiling .pio/build/mhetesp32minikit/lib3e6/ESP Async WebServer_ID306/WebResponses.cpp.o
Compiling .pio/build/mhetesp32minikit/lib3e6/ESP Async WebServer_ID306/WebServer.cpp.o
Archiving .pio/build/mhetesp32minikit/libFrameworkArduinoVariant.a
Indexing .pio/build/mhetesp32minikit/libFrameworkArduinoVariant.a
Compiling .pio/build/mhetesp32minikit/FrameworkArduino/Esp.cpp.o
Compiling .pio/build/mhetesp32minikit/FrameworkArduino/FunctionalInterrupt.cpp.o
Compiling .pio/build/mhetesp32minikit/FrameworkArduino/HardwareSerial.cpp.o
Archiving .pio/build/mhetesp32minikit/lib3e6/libESP Async WebServer_ID306.a
[... 28 lines cut ...]
Compiling .pio/build/mhetesp32minikit/FrameworkArduino/esp32-hal-rmt.c.o
Compiling .pio/build/mhetesp32minikit/FrameworkArduino/esp32-hal-sigmadelta.c.o
Compiling .pio/build/mhetesp32minikit/FrameworkArduino/esp32-hal-spi.c.o
Compiling .pio/build/mhetesp32minikit/FrameworkArduino/esp32-hal-time.c.o
Compiling .pio/build/mhetesp32minikit/FrameworkArduino/esp32-hal-timer.c.o
Compiling .pio/build/mhetesp32minikit/FrameworkArduino/esp32-hal-touch.c.o
Compiling .pio/build/mhetesp32minikit/FrameworkArduino/esp32-hal-uart.c.o
Compiling .pio/build/mhetesp32minikit/FrameworkArduino/libb64/cdecode.c.o
Compiling .pio/build/mhetesp32minikit/FrameworkArduino/libb64/cencode.c.o
Compiling .pio/build/mhetesp32minikit/FrameworkArduino/main.cpp.o
Compiling .pio/build/mhetesp32minikit/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/mhetesp32minikit/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/mhetesp32minikit/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/mhetesp32minikit/libFrameworkArduino.a
Indexing .pio/build/mhetesp32minikit/libFrameworkArduino.a
[... cut ...]
Retrieving maximum program size .pio/build/mhetesp32minikit/firmware.elf
Building .pio/build/mhetesp32minikit/firmware.bin
Checking size .pio/build/mhetesp32minikit/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=         ]  11.7% (used 38448 bytes from 327680 bytes)
PROGRAM: [======    ]  55.1% (used 722106 bytes from 1310720 bytes)
esptool.py v2.6
=============================================================== [SUCCESS] Took 24.46 seconds ===============================================================
```

atanisoft, 2019-06-30:
> Can you provide a Minimal Complete Reproducible Example which demonstrates this?

I just retested with 1.9.0 and can no longer reproduce the failure. Still not entirely clear why it was a problem previously. I'm still using the #feature/stage version as I have dependencies on the 1.0.3-rc1 code (in the process of shifting to IDF instead of Arduino APIs).

waltsteinchen, 2019-07-01:
I have the same problem using latest Arduino IDE 1.8.9: 
c:\Users\J\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:201:141: error: 'xTaskCreateUniversal' was not declared in this scope

xTaskCreateUniversal(_async_service_task, "async_tcp", 8192 * 2, NULL, 3, &_async_service_task_handle, CONFIG_ASYNC_TCP_RUNNING_CORE);

Any workaround available here ?

atanisoft, 2019-07-01:
@waltsteinchen make sure you are using the arduino-esp32 1.0.2 release (or later) in the Arduino IDE. It is entirely possible that you have a cached copy of the 1.0.1 release which the Arduino IDE is still using instead of the expected 1.0.2 (or 1.0.3-rc1) code.

waltsteinchen, 2019-07-02:
After a few unsuccessful attempts I did a complete new install in a VM of everything .. and indeed it works there .. so I'll hunt where the mess is now  ..
Thanks for your ultrafast help!

jptrsn, 2019-07-02:
Updating to the latest platformio build has resolved the issue for me.

closed 2019-07-02 by jptrsn