me-no-dev/AsyncTCP #52: How to restart ESP whenever I get this : "[W][AsyncTCP.cpp:896] _poll(): ack timeout 4" ?

GeorgeFlorian, 2019-07-09:
Board: ESP32

Hello !
As per-title I would like to know if it's possible to restart the ESP whenever I get the following warning:
```
[W][AsyncTCP.cpp:896] _poll(): ack timeout 4
```
This basically means that the server timed-out and it can't be used. And the only method of restoring it is to manually reset the ESP.

bertmelis, 2019-07-09:
It means that the server you're connecting to didn't send an ack in time but the ESP is perfectly capable of handling this error. But you have to implement it offcourse (use the timeout callback, close connection and try to reopen?)

GeorgeFlorian, 2019-07-09:
> But you have to implement it offcourse (use the timeout callback, close connection and try to reopen?)

If you don't mind me asking, how can I do that ?

bertmelis, 2019-07-12:
```C++
client.onTimeout([](void*, AsyncClient*, uint32_t time) {
  client.close(true);  // force close connection

  // try connect immediately
  client.connect(ipAddress, pot);
  // or set a timer and connect on fire
 ticker.once(2, connectFunction);
});
```

GeorgeFlorian, 2019-07-15:
> ```c++
> client.onTimeout([](void*, AsyncClient*, uint32_t time) {
>   client.close(true);  // force close connection
> 
>   // try connect immediately
>   client.connect(ipAddress, pot);
>   // or set a timer and connect on fire
>  ticker.once(2, connectFunction);
> });
> ```
> 
> (edit) It involves more code than shown, but I hope you get the idea?

I don't know what `client` is. It is a socket client as in `AsyncWebSocketClient * client;` ?

The only Async structure I use is this: `AsyncWebServer server(80);`.
I don't use these :
```
AsyncWebSocket ws("/ws");
AsyncEventSource events("/events");
```
and I don't believe I have access to the `client` object.

GeorgeFlorian, 2019-07-15:
I can't even do this: `Serial.println(client->getLocalAddress());` properly.
All I did was to add `AsyncClient *client;` outside `void setup()` and the `println()` at the end of the setup.
It returns: `Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.`

How do you actually use this ?

GeorgeFlorian, 2019-07-15:
I would really like to know how to use `client.onTimeout([](void*, AsyncClient*, uint32_t time) {});` , because I get very few `ack timeout 4` freezes so an `ESP.restart()` can work. But I get fairly more `rx timeout 4` and very few of them actually freeze the server, so I can't restart the ESP that frequently because I won't be able to do a thing on the server.

bertmelis, 2019-07-15:
Well, I would have to dig deeper: as you posted this issue in the TCP lib instead of the webserver lib I had no clue it was about the  webserver actually.

cliënt refers to the asnycclient but that is unavailable in the webserver library.

GeorgeFlorian, 2019-07-15:
> Well, I would have to dig deeper: as you posted this issue in the TCP lib instead of the webserver lib I had no clue it was about the webserver actually.
> 
> cliënt refers to the asnycclient but that is unavailable in the webserver library.

I opened the issue here because the warning messages come from the AsyncTCP library.

I have opened an issue there as well: https://github.com/me-no-dev/ESPAsyncWebServer/issues/553

dao89, 2019-07-20:
I am using AsyncTCP and AsyncMQTT with ESP32. After some time even ping to ESP32 fails, but it stil connected to wifi router. Only thing in log is "_poll(): ack timeout 4". Disconnecting ESP32 from WIFi (via router's interface) bring things to work after some seconds. So I believe than AsyncTCP has some bug, It is impossible to use it now...

closed 2019-10-05 by stale[bot]

brodrigz, 2024-10-08:
This bug is very present in the latest version

shacal, 2025-01-06:
Hosting http server,

`[ 63107][W][AsyncTCP.cpp:1001] _poll(): ack timeout 4
[186135][W][AsyncTCP.cpp:1001] _poll(): ack timeout 4
[243135][W][AsyncTCP.cpp:1001] _poll(): ack timeout 4
[273135][W][AsyncTCP.cpp:1001] _poll(): ack timeout 4
[336135][W][AsyncTCP.cpp:1001] _poll(): ack timeout 4
[544137][W][AsyncTCP.cpp:1001] _poll(): ack timeout 4`