me-no-dev/AsyncTCP #54: Serial console flooded with [E][AsyncTCP.cpp:897] _poll(): messages

tom80000, 2019-07-15:
Hi!

My first post on Github! Yay?

I started using AsyncTCP in combination with AsyncMqttClient (https://github.com/marvinroger/async-mqtt-client) and its working really good. No connectivity issues whatsoever. Except that my serial console is flooded with error messages:

[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[WiFi-event] event: 0
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 4 - STA_CONNECTED
[WiFi-event] event: 4
[WiFi-event] event: 4
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 10.0.2.191, MASK: 255.255.255.0, GW: 10.0.2.254
10.0.2.191
Connecting to MQTT...
[WiFi-event] event: 7
WiFi connected
IP address: 
10.0.2.191
Connecting to MQTT...
Connected to MQTT.
Session present: 0
[E][AsyncTCP.cpp:832] _lwip_fin(): 0x3ffba78c != 0x3ffcc6ec
[E][AsyncTCP.cpp:897] _poll(): 0x3ffba78c != 0x3ffcc6ec
[E][AsyncTCP.cpp:897] _poll(): 0x3ffba78c != 0x3ffcc6ec
[E][AsyncTCP.cpp:897] _poll(): 0x3ffba78c != 0x3ffcc6ec
[E][AsyncTCP.cpp:897] _poll(): 0x3ffba78c != 0x3ffcc6ec
Connecting to MQTT...
[W][AsyncTCP.cpp:650] connect(): already connected, state 4
[E][AsyncTCP.cpp:897] _poll(): 0x3ffba78c != 0x3ffcc6ec
[E][AsyncTCP.cpp:897] _poll(): 0x3ffba78c != 0x3ffcc6ec
[E][AsyncTCP.cpp:897] _poll(): 0x3ffba78c != 0x3ffcc6ec
[E][AsyncTCP.cpp:897] _poll(): 0x3ffba78c != 0x3ffcc6ec
....... continues forever......

Can someone please explain what is going on? What am I doing wrong?

Apparently esp_log_level_set does not work in the Arduino environment? So I can't ignore the message. Completely disabling logging is not an option for me. I could remove the log_e line in AsyncTCP.cpp but I rather keep the libraries as is.

My setup:
ESP32-wrover
Mikrotik Wifi
Mosquitto MQTT server/broker
Arduino environment with visual studio code.

gabrielrbeiro, 2019-07-29:
Same problem here with messages like: 

[E][AsyncTCP.cpp:721] _poll(): 0x3ffde458 != 0x3ffdefa8
[E][AsyncTCP.cpp:721] _poll(): 0x3ffdee38 != 0x3ffdefa8
[E][AsyncTCP.cpp:721] _poll(): 0x3ffdecc8 != 0x3ffdefa8
[E][AsyncTCP.cpp:721] _poll(): 0x3ffdeb58 != 0x3ffdefa8
[E][AsyncTCP.cpp:721] _poll(): 0x3ffde878 != 0x3ffdefa8
[E][AsyncTCP.cpp:721] _poll(): 0x3ffde9e8 != 0x3ffdefa8
[E][AsyncTCP.cpp:721] _poll(): 0x3ffde708 != 0x3ffdefa8

how to disable ?

bertmelis, 2019-07-29:
You should not disable and something's wrong. asyncmqtt relies on the onPoll callback so I doubt if everything keeps working.

tom80000, 2019-07-29:
> Same problem here with messages like:
>
> [E][AsyncTCP.cpp:721] _poll(): 0x3ffde458 != 0x3ffdefa8
> [E][AsyncTCP.cpp:721] _poll(): 0x3ffdee38 != 0x3ffdefa8
> [E][AsyncTCP.cpp:721] _poll(): 0x3ffdecc8 != [...]

You could comment out (in your case) line 721 in AsyncTCP.cpp to get rid of the error message but that doesn't solve the actual problem.

Unfortunately I don't know how protocol control blocks work so I don't know how to fix the real problem.

closed 2019-10-11 by stale[bot]

CircuitSetup, 2020-05-12:
Has anyone ever figured anything out with this? I'm seeing the same issue. It appears to be related to the handling of _pull.

bluelu, 2020-05-20:
I also have the same issue

proddy, 2020-05-24:
me too. but not all the time. it's odd.

BlueAndi, 2020-05-24:
The AsyncClient gets a invalid protocol control unit from the lwip stack. Its interesting in which case this starts happening. If you comment in all other debug logs, it may help to figure it out.

The client instance is determined by argument: https://github.com/me-no-dev/AsyncTCP/blob/ca8ac5f919d02bea07b474531981ddbfd64de97c/src/AsyncTCP.cpp#L1205-L1207

The argument is set via tcp_arg().

Are you using the client for several connections? Are you assinging clients? Does it happend during a disconnect/reconnt?

rowdy15, 2021-03-06:
Try changing the wifi name that the esp32 is broadcasting. This worked for me.

WiFi.setHostname("wifi_name"); // change this name

The cause:

You may have programmed 2 different devices with the same broadcasted wifi name and so the router in the network doesn't like it when there are 2 different mac addresses with the same wifi name.

I don't know if this the reason for sure, but certainly changing the wifi name of the device made it go away for me.

DTTerastar, 2021-03-25:
I'm getting this on just ONE of my m5stack-atoms...  4 others work great...

slaesh, 2023-01-30:
same issue here.

> Are you using the client for several connections? Are you assinging clients? Does it happend during a disconnect/reconnt?

yes it does appear after a disconnect/reconnect. but not that often that I could debug it properly :(

schuppeste, 2023-11-05:
Here too.. in my Case i send Post Data.

after i send data to /postconfig i cant access my /getconfig (ajax) to fill out forms.

But i can send /postconfig every time new, it works, index.html too.