me-no-dev/AsyncTCP #55: Changing priority of task

AckmaniacSoftware, 2019-07-15:
I tried to port my code from ESP8266 to ESP32 and became timing issues. Problem was that on the ESP8266 the requests were handled in the loop and not Async at any time. This caused timing issues in the rest of the logic which is critical for my application. I tried to fix that by suspending the AsyncTCP task but it didn't work exactly how i wanted.
Finally a hige difference made the change of the tasks priority.

Changing priority from 3
`xTaskCreateUniversal(_async_service_task, "async_tcp", 8192 * 2, NULL, 3, &_async_service_task_handle, CONFIG_ASYNC_TCP_RUNNING_CORE);`

to 1 fixed all my issues
`xTaskCreateUniversal(_async_service_task, "async_tcp", 8192 * 2, NULL, 1, &_async_service_task_handle, CONFIG_ASYNC_TCP_RUNNING_CORE);`

Now my code runs reliable without any noticeable impact. 

So is there a reason that by default the priority is set to 3?

atanisoft, 2019-07-15:
It runs at 3 so it has a higher priority than the loop() method. It should have no impact on your code as long as you don't abuse the callbacks (ie run something that isn't directly related to sending/receiving data)

AckmaniacSoftware, 2019-07-15:
I work with shit registers and need to switch their state as fast as possible. This is done directly in the software via switching a PIN state from HIGH to LOW and vice versa. And setting the priority to 1 fixed the issue that request handling had an effect on the switching times. And the requests are still handled perfectly.

closed 2019-10-05 by stale[bot]