me-no-dev/AsyncTCP #71: Support for ESP-IDF

kaizoku-619, 2019-10-06:
Is there any support for AsyncTCP or ESPAsyncWebServer in ESP-IDF? if no can we use a C++ library like this in ESP-IDF?

me-no-dev, 2019-10-06:
you can if you install Arduino as IDF component and you use IDF 3.2 (the version that Arduino currently uses). if you want to try it, you can find instructions in the readme of arduino for esp32 :)

me-no-dev, 2019-10-06:
Actually I went through the code and it will not be hard to run AsyncTCP with IDF only, but for the web server, that will be a different story. It has links to a lot of Arduino classes. Maybe some core part of it could be extracted, but a lot of the handlers will need to be rewritten.

closed 2019-10-06 by me-no-dev