me-no-dev/AsyncTCP #74: ESP32 rebooting when file serving from

d0ct0rd-dalimo, 2019-10-28:


amrlsayed, 2019-11-07:
I have the same problem

amrlsayed, 2019-11-07:
same problem with the ESP_AsyncFSBrowser example with big webpages.
but when I minimize the loop code it works much stable. so I changed the loop of the example to 
```
void loop(){
  //ArduinoOTA.handle();
  ws.cleanupClients();
  delay(1);
}
```
commenting out the OTA code.
it seems related to this comment https://github.com/me-no-dev/AsyncTCP/issues/69#issuecomment-537414042

amrlsayed, 2019-11-09:
I have solved my problem, by putting all the CSS files into a single one and same for the JS files. this helped minimizing the number of requests handles at the same time by the ESP32. 
So for my case it's not related if the web pages is small or large but related to the number of request handled at the same time.

closed 2020-01-22 by stale[bot]