me-no-dev/AsyncTCP #76: AsyncTCP.cpp:221:141: error: 'xTaskCreateUniversal' was not declared

MartinDavidWaller, 2019-12-09:
Hi,

I have an ESP32 project that I've just tried to rebuild using the latest Arduino IDE 1.8.10 and the latest down load of the AyncTCP project. The first error I get from the compiler is:

:\Users\marti\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp: In function 'bool _start_async_task()':

C:\Users\marti\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:221:141: error: 'xTaskCreateUniversal' was not declared in this scope

         xTaskCreateUniversal(_async_service_task, "async_tcp", 8192 * 2, NULL, 3, &_async_service_task_handle, CONFIG_ASYNC_TCP_RUNNING_CORE);

I wonder if you can help me please?

Martin

Bmooij, 2019-12-16:
Which version of the ESP32 Arduino are you using?

MartinDavidWaller, 2019-12-17:
Hi, 

Many thanks for getting back to me.

I'm using the Arduino IDE 1.8.10

For the ESP32 board manager I'm using the URL: 

http://dl.espressif.com/dl/package_esp32_index.json

I'm using the AsyncTCP in collaberation with ESP Async WebServer V1.2.3

The actual hardware being used is one one these:

https://www.ebay.co.uk/itm/ESP32-Development-Board-Wireless-WiFi-Bluetooth-2-in1-Dual-Core-for-CP2102-ESP32/153034283929?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

Does that help? Am I missing another version number that could help you?

Martin

Bmooij, 2019-12-17:
Hi Martin,

In the Arduino boardmanager you find the ESP32 Arduino version. It is something like 1.0.x

MartinDavidWaller, 2019-12-17:
Hi Bmooij,

Sorry, I knew I'd missed one. Yes, it's esp32 by Espressif Systems version 1.0.1

Does that help?

Martin

Bmooij, 2019-12-17:
Hi Marin,

Yes, the function xTaskCreateUniversal is introduced after 1.0.1.
Please update your board to the latest version (1.0.4).

MartinDavidWaller, 2019-12-17:
Hi Bob,

That sorted it. It now compiles without issue. Many thanks for all your help with this.

Martin

closed 2019-12-17 by MartinDavidWaller

mike-one, 2020-07-04:
you can replace it with "xTaskCreatePinnedToCore"
Like this: xTaskCreatePinnedToCore(_async_service_task, "async_tcp", 8192 * 2, NULL, 3, &_async_service_task_handle, CONFIG_ASYNC_TCP_RUNNING_CORE);