me-no-dev/AsyncTCP #78: ESPAsync stops working after AsyncTCP.cpp warning (timeout)

ViniP8, 2019-12-17:
Hi!

From time to time, the ESPAsync server slows down, freezes for a few seconds, or completely stops working until reboot (or until the user disconnects and reconnects to the AP).

Enabling verbose mode at serial monitor, some warnings show up before the problem happens (not all of them at the same time, sometimes only one of these happens):

`[W][AsyncTCP.cpp:893] _poll(): pcb is NULL`
`[W][AsyncTCP.cpp:913] _poll(): rx timeout 4`
`[W][AsyncTCP.cpp:906] _poll(): ack timeout 4`

Is there any way to check if this problems has happened, so that I can restart the server?

This error usually happens when loading more than one file at time (>10KB) with the ESP on AP mode. 

Tested with Chrome, Firefox, running on Linux (ubuntu) and Windows.

zhivko, 2019-12-18:
Try in your client:

if (ws.hasClient(lastWsClient)) {
if (ws.availableForWrite(lastWsClient))
ws.text(lastWsClient, txtToSend);
}

I know sometimes network/browser results in browser unable to receive data,
that data then piles up inside asynctcp....

Also heap grows in server if this happens.

Also add call to this function in separate task every 10seconds:
It helped me with stucked tcpip stack of esp-idf...

void forceARP()
{
char *netif = netif_list;
while(netif)
{
netif = *((char **) netif);
}
}

Report if you had succes with this.

After this modifications, my project based on asynwebserver that uses we
sockets, runs now for:

106.12 hours

And is stil responsive.

ViniP8, 2019-12-18:
Thank you very much!!

I've never used the WebSocket plugin, so i'll try to use it and report if the solutoin works!!

Just one question: The variable "netif_list" from the forceARP() function comes from where? (I'm using arduino IDE and didn't found it)

zhivko, 2019-12-20:
WS client should be part of javacript code, that "resides" in your html, that you serve from your esp32 http server.
This html is then returned to browser(client), and js code is executed. In that moment websocket client is created that sends or/and retrieves data from esp32 asyncwebserver.

ViniP8, 2019-12-20:
Got the idea!!

The problem is that sometimes while going from one page to another the server stops (before loading it)

For now, I'm using a ping library to ping my gateway, because apparently when the server stops, the esp can't even ping the router. When this happens, I'm disconnecting and reconnecting from the network. 

Of course that pinging the gateway and disconnecting from the network isn't a good solution but works 

I'll study the ESP library for arduino to see if there is any way to clear the TCP/Ip stack

ViniP8, 2019-12-20:
Using the following code, i'm able to compile the code using the `forceARP()` function, but the problem continues happening (server freezing after ack timeout):
```
extern "C" {
   char *netif_list;
   uint8_t etharp_request(char *, char *);
 }

void forceARP(){
    char *netif = netif_list;
    while(netif){
        netif = *((char **) netif);
    }
}
```

I tried removing the `uint8_t etharp_request(char *, char *);`, but the problem continues. 

Then I searched on internet about the `etharp_request`, but using `etharp_request((netif), (netif+4));` (used in a few solutions for ESP8266) results in:

> assertion "netif->hwaddr_len must be the same as ETH_HWADDR_LEN for etharp!" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/netif/etharp.c" 

(maybe because i'm using ESP32?)

Any other ideas?

From now and on thank you very much!

closed 2020-03-03 by stale[bot]

GeorgeFlorian, 2020-08-06:
Any updates on this ? It's still happening.