me-no-dev/AsyncTCP #79: cannot set CONFIG_ASYNC_TCP_RUNNING_CORE from outside

alorbach, 2020-01-07:
Maybe I am to stupid today, but If I define CONFIG_ASYNC_TCP_RUNNING_CORE before including AsyncTCP.h, it does not have any affect at all: 
```
# define CONFIG_ASYNC_TCP_RUNNING_CORE 0
# include <AsyncTCP.h>

```

Itz still runs into the ifndef block in AsyncTCP.h
```
#ifndef CONFIG_ASYNC_TCP_RUNNING_CORE
#define CONFIG_ASYNC_TCP_RUNNING_CORE -1 //any available core
#define CONFIG_ASYNC_TCP_USE_WDT 1 //if enabled, adds between 33us and 200us per event
#endif

```

andrea-daru, 2020-01-16:
I've the same problem.. @alorbach have you find a solution? The only workaround I've found it's to modify the library

alorbach, 2020-01-17:
Yes that is what I did finally, i modified CONFIG_ASYNC_TCP_RUNNING_CORE default in AsyncTCP.h

bertmelis, 2020-01-17:
Use platformio to build your "sketches" and set a build flag like it's done here: https://docs.platformio.org/en/latest/platforms/espressif32.html#debug-level

alorbach, 2020-01-17:
Does this work with Arduino IDE?

bertmelis, 2020-01-18:
no

boblemaire, 2020-02-27:
@bertmelis, Like everyone else, I had a modified asynctcp.h, but adding:
build_flags = -DCONFIG_ASYNC_TCP_RUNNING_CORE 0
causes this error for me:
*** [.pio\build\esp32dev\firmware.elf] Implicit dependency `C:\Users\Bob\.platformio\platforms\espressif32\builder\0' not found, needed by target `.pio\build\esp32dev\firmware.elf'.
================
I like this idea, and have another place that could use it as well.  Any idea what is wrong here?

closed 2020-05-11 by stale[bot]