me-no-dev/AsyncTCP #80: wdt not reset by async tcp

robertgregor, 2020-02-14:
2020-02-14 15:07:34 V (44119) detAndCancelInterrupt: Detached interrupt from ALARM_PIN and set the interruptFired = false
2020-02-14 15:07:34 I (44121) RemoteHome::registerWebServerHandler: WebServer: datatables.min.js requested.
2020-02-14 15:07:34 V (44445) atInterrupt: Attached RISING interrupt to ALARM_PIN
E (49565) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (49565) task_wdt:  - async_tcp (CPU 0/1)
E (49565) task_wdt: Tasks currently running:
E (49565) task_wdt: CPU 0: IDLE0
E (49565) task_wdt: CPU 1: IDLE1
E (49565) task_wdt: Aborting.
abort() was called at PC 0x401038dc on core 0

Backtrace: 0x4008cce4:0x3ffbe160 0x4008cf15:0x3ffbe180 0x401038dc:0x3ffbe1a0 0x40081c2d:0x3ffbe1c0 0x401897cf:0x3ffbc1e0 0x4010176a:0x3ffbc200 0x4008ab89:0x3ffbc220 0x40088c89:0x3ffbc240

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8740
load:0x40080400,len:5788
entry 0x4008069c
1970-01-01 00:00:00 I (204) RemoteHome::initSpiffFiles: Read the out counter value 0 from file /outcounter.
1970-01-01 00:00:00 I (206) RemoteHome::initSpiffFiles: Read the in counter value 0 from file /incounter.
1970-01-01 00:00:00 I (212) RemoteHome::setup: Starting module...

atanisoft, 2020-02-14:
Without decoding the backtrace *AND* providing code to reproduce this, it won't likely be worked on at all.

XYZVector, 2020-02-15:
Yea you need a lot more information otherwise nobody knows how to reproduce it or what is, and your exception occurred in the 1970's LOL :).

robertgregor, 2020-02-17:
Backtrace: 0x4008cce4:0x3ffbe160 0x4008cf15:0x3ffbe180 0x401038dc:0x3ffbe1a0 0x40081c2d:0x3ffbe1c0 0x401897cf:0x3ffbc1e0 0x4010176a:0x3ffbc200 0x4008ab89:0x3ffbc220 0x40088c89:0x3ffbc240

is decoded:

0x4008cce4: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x4008cf15: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x401038dc: esp_read_mac at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/system_api.c line 198
0x401897cf: reverse at C:\Users\gregorro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3\cores\esp32\stdlib_noniso.c line 32
0x4010176a: esp_event_set_default_wifi_handlers at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/event_default_handlers.c line 418
0x4008ab89: prvIdleTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 3382
0x40088c89: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

So it seems is inside idf. I am going to close this then...

closed 2020-02-17 by robertgregor