me-no-dev/AsyncTCP #81: fatal error: interrupts.h: No such file or directory

m-menon, 2020-02-15:
I get this error when I try to compile on PlatformIO. The framework is Arduino. Is there any known fix for this?

```
In file included from .pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\SyncClient.h:32:0,
                 from .pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\SyncClient.cpp:22:
.pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src/async_config.h:13:0: note: this is the location of the previous definition
 #define TCP_MSS (1460)
 ^
.pio\libdeps\esp32dev\ESPAsyncTCP_ID305\src\SyncClient.cpp:25:24: fatal error: interrupts.h: No such file or directory

********************************************************************
* Looking for interrupts.h dependency? Check our library registry!
```

atanisoft, 2020-02-15:
@m-menon The latest PlatformIO Core broke LDF it seems and your code is trying to compile both the ESP32 and ESP8266 versions. The above compilation failure can be fixed with this addition to your platformio.ini file:
```
lib_compat_mode=strict
```
With this flag LDF ignores libraries which do not apply to the current platform (among other checks).

m-menon, 2020-02-15:
@atanisoft okay cool. I'll try this method as well.

mvadu, 2020-02-16:
Thank you @atanisoft It worked like magic.. I spent hours re-installing the platformio IDE, deleting cached libs etc until I saw your suggestion.

closed 2020-04-30 by stale[bot]

Wafiifaw55, 2021-10-05:
@atanisoft Hi dear I have same problem, but I don't understand the solution you tell us, can you explain the steps please?