me-no-dev/AsyncTCP #82: Bind error: -8

spyder0069, 2020-02-18:
Every so often in my project I get this when the server tries to begin:

[E][AsyncTCP.cpp:1287] begin(): bind error: -8

The device has connected to the network and received an IP at this point.  I can't seem to find info on what this is.

spyder0069, 2020-02-18:
I should add this is a esp32 wroom and it happens when the async webserver tries to start.  At that point the server doesn't start.  If I do a server.end(); and try again I usually get the same error.   The error shows in the serial debug but I don't see a way to check in my code what the status of the webserver is and if it successfully started.

MiLeG, 2020-03-15:
Yep, getting the same issue here.
Happens to me when I try restarting the webserver (end() and then begin()), but not always.

BlueAndi, 2020-04-09:
Error "-8" comes from lwip tcp_bind().
```
/** Address in use.          */
  ERR_USE        = -8,
```

closed 2020-06-23 by stale[bot]

yossilan, 2020-07-20:
I'm experiencing the same issue. Any update/info on the cause?

spyder0069, 2020-07-20:
In my project I removed the server.end calls.  Once its started I just leave it alone.

vilakshan2996, 2023-03-01:
Is there any solutions now??

XDGFX, 2023-03-15:
I'm having the same issue. A workaround seems to be to never call `server.end()` but this seems to introduce additional errors (i.e. if switching WiFi networks, or switching from STA to AP or vice versa).

It would be good if this was fixed so `server.end()` was usable.