me-no-dev/AsyncTCP #84: problem using any async libs (async webserver or even mqtt) with lora

AliBigdeli, 2020-02-28:
for a while i was working on a project as a packet forwarder for my esp32 and it was working pretty good but when i switched it to async i found out a real problem, like when i get any request to send a lora packet it sends a few and stops sending (after like 3 to 9).
its actually because of async i have no idea why but i kept it simple like a web page to open and send packet before response to client and it fails, its not happening on httpserver and pubsub but only on everything with async lib.

trycoon, 2020-02-28:
I really can't help you, but usually the guys here would ask for some source code to further assist you.
But out of curiosity, what LoRa-library are you using? I'm just starting up a project where I need a ESP32 to talk to a Raspberry Pi over LoRa (not LoRa-WAN), so it would be nice with some guidance of a suitable LoRa-library. I'm using SX1280 devices. Thanks!

AliBigdeli, 2020-03-01:
> I really can't help you, but usually the guys here would ask for some source code to further assist you.
> But out of curiosity, what LoRa-library are you using? I'm just starting up a project where I [...]

AliBigdeli, 2020-03-01:
well here is a demo of problem , use the code below to send message with a request to this url YOURIP/hello it will send a message through lora you can use lora duplex example for the other side to see the problem, in code it shows that it is sending and no error or even watch dog trigger or anything else but in real nothing will work after send a few and even recieving side will stop working too, (i clear this once more only on anything async this problem is happening not others)

#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <LoRa.h>
#include <SPI.h>
AsyncWebServer server(80);
//AsyncWebSocket ws("/ws");

String outgoing;              // outgoing message

byte msgCount = 0;            // count of outgoing messages
byte localAddress = 0xBB;     // address of this device
byte destination = 0xFF;      // destination to send to
long lastSendTime = 0;        // last send time
int interval = 2000;          // interval between sends

const char* ssid = "BlackFox";
const char* password = "a/@1234567";

void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, password);
 
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi..");
  }
 
  Serial.println(WiFi.localIP());
 
  server.on("/hello", HTTP_GET, [](AsyncWebServerRequest *request){
    String message = "HeLoRa World! web";   // send a message
    sendMessage(message);
    request->send(200, "text/plain", "Hello World");
  });
  
  while (!Serial);
  
  Serial.println("LoRa Duplex");

  // override the default CS, reset, and IRQ pins (optional)
  LoRa.setPins(13, 17, -1);// set CS, reset, IRQ pin

  if (!LoRa.begin(433E6)) {             // initialize ratio at 915 MHz
    Serial.println("LoRa init failed. Check your connections.");
    while (true);                       // if failed, do nothing
  }
  server.begin();
  Serial.println("LoRa init succeeded.");
}

void loop() {
  if (millis() - lastSendTime > interval) {
    String message = "HeLoRa World!";   // send a message
    sendMessage(message);
    Serial.println("Sending " + message);
    lastSendTime = millis();            // timestamp the message
    interval = random(2000) + 1000;    // 2-3 seconds
  }

  // parse for a packet, and call onReceive with the result:
  onReceive(LoRa.parsePacket());
}

void sendMessage(String outgoing) {
  LoRa.beginPacket();                   // start packet
  LoRa.write(destination);              // add destination address
  LoRa.write(localAddress);             // add sender address
  LoRa.write(msgCount);                 // add message ID
  LoRa.write(outgoing.length());        // add payload length
  LoRa.print(outgoing);                 // add payload
  LoRa.endPacket();                     // finish packet and send it
  msgCount++;                           // increment message ID
}

void onReceive(int packetSize) {
  if (packetSize == 0) return;          // if there's no packet, return

  // read packet header bytes:
  int recipient = LoRa.read();          // recipient address
  byte sender = LoRa.read();            // sender address
  byte incomingMsgId = LoRa.read();     // incoming msg ID
  byte incomingLength = LoRa.read();    // incoming msg length

  String incoming = "";

  while (LoRa.available()) {
    incoming += (char)LoRa.read();
  }

  if (incomingLength != incoming.length()) {   // check length for error
    Serial.println("error: message length does not match length");
    return;                             // skip rest of function
  }

  // if the recipient isn't this device or broadcast,
  if (recipient != localAddress && recipient != 0xFF) {
    Serial.println("This message is not for me.");
    return;                             // skip rest of function
  }

  // if message is for this device, or broadcast, print details:
  Serial.println("Received from: 0x" + String(sender, HEX));
  Serial.println("Sent to: 0x" + String(recipient, HEX));
  Serial.println("Message ID: " + String(incomingMsgId));
  Serial.println("Message length: " + String(incomingLength));
  Serial.println("Message: " + incoming);
  Serial.println("RSSI: " + String(LoRa.packetRssi()));
  Serial.println("Snr: " + String(LoRa.packetSnr()));
  Serial.println();
}

AliBigdeli, 2020-03-01:
> I really can't help you, but usually the guys here would ask for some source code to further assist you.
> But out of curiosity, what LoRa-library are you using? I'm just starting up a project where I [...]

well ill be glad to help you out cause we have built a platform for lora communication (software and hardware) which will be coming out soon but i can guide ya through your project contact me by my mail

trycoon, 2020-03-01:
Sounds interesting, would be nice with a PM once you have something working. Just to see if it fits my needs.

AliBigdeli, 2020-03-02:
there is something weird about it cause when i use a flag in loop to send the message it works totally fine but when i want to call the function directly it doesnt work, ill be using this method for now but still there is a problem that needs fixing

closed 2020-05-15 by stale[bot]