me-no-dev/AsyncTCP #88: ESP32 crashes while connecting to server ... sometimes

joaolrc, 2020-04-14:
Hi All. So i wrote a library for Async HTTP calls on top of AsyncTCP and i am testing , making 2 concurrent requests and sometimes i get this error followed by a crash:

```
assertion "new_rcv_ann_wnd <= 0xffff" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c", line 779, function: tcp_update_rcv_ann_wnd
abort() was called at PC 0x40118acf on core 0

Backtrace: 0x4008c454:0x3ffb3f80 0x4008c685:0x3ffb3fa0 0x40118acf:0x3ffb3fc0 0x40132b32:0x3ffb3ff0 0x40132bc1:0x3ffb4010 0x4016a4ba:0x3ffb4030 0x4012f07c:0x3ffb4050 0x40088b9d:0x3ffb4080

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
```

## Header File (AsyncHttpClient.h)

```c++
#ifndef ASYNC_HTTP_CLIENT_H
#define ASYNC_HTTP_CLIENT_H

#include <Arduino.h>
#include <functional>
#include <AsyncTCP.h>

// #define AHC_VERBOSE_LVL 3 //0 to 3 --> adjust to modify verbose
#define AHC_MAX_RESPONSE_SIZE 5000
#define AHC_MAX_REQBODY_SIZE 1000
#define AHC_EMPTY_RESPONSE -1

// typedef std::function<void(int, String)> ahc_responseCallback_t;
typedef void (*ahc_responseCallbackPtr_t)(int,String);      //pointer to request callback type

//to store response from server parsed
typedef struct {
    int httpCode;
    String responseBody;
}ahc_requestResponse_t;

typedef struct {
    String host;
    String query;
    String body;
    ahc_responseCallbackPtr_t respCallbackPtr;
    uint16_t port;
    bool islastMethodGet;
}ahc_lastCalledParameters_t;

class AsyncHttpClient{
public:

    bool get( String host, String query="", ahc_responseCallbackPtr_t callback = NULL, uint16_t port=80 );
    bool post( String host, String query="", String body="", ahc_responseCallbackPtr_t callback = NULL, uint16_t port=80 );
    bool cancelRequest();
    

private:

    AsyncClient * _aClient = NULL;
    String _dataToSend = "";
    String _lastServerResponse = "";
    ahc_lastCalledParameters_t _myLastParameters= {"", "", "", NULL, 80, true};
    uint8_t _connectionAttemptCounter = 0;
[... 242 lines cut ...]
bool AsyncHttpClient::cancelRequest(){
    if (_aClient==NULL){
        return false;
    }
    if (_aClient->connected() && _lastServerResponse=="" ){
        debugPrint(F("[AsyncHttpClient] Canceling Request. Closing open connenction\n"), 1);
        _aClient->close();
    }
    if (_aClient->connecting() ){
        debugPrint(F("[AsyncHttpClient] Canceling Request. Aborting connection request\n"), 1);
        _aClient->close();
    }
    return true;
}

void AsyncHttpClient::debugPrint(String stringToPrint, uint8_t verboseLvl){
#ifdef AHC_VERBOSE_LVL
    if (AHC_VERBOSE_LVL >= verboseLvl)
        Serial.print(stringToPrint);
#endif
}
```

## Testing Notes
I am instantiating the class twice and making concurrent requests in a for loop.The returned response is JSON. I think my problem is how i handle deletion and heap allocation of the AsybcTCP client instance maybe in the get and post methods but im fighting this battle for some days now and starting to get a bit frustrated. Can someone help me on this?

closed 2020-06-27 by stale[bot]

ivancmz, 2020-07-15:
I'm having exactly the same problem, did you find anything @joaolrc

joaolrc, 2020-07-15:
@ivancmz not really. I just started using this library https://github.com/boblemaire/asyncHTTPrequest
Works well and for now it's ok byut still want to finish my own. Another cool thing i wish to have is https connections. Do you know how can i achieve this?

BlueAndi, 2020-07-18:
Decode the backtrace, otherwise its difficult.

Anyway I did the same ;-) and its working:
https://github.com/BlueAndi/esp-rgb-led-matrix/blob/master/src/Web/AsyncHttpClient.h
https://github.com/BlueAndi/esp-rgb-led-matrix/blob/master/src/Web/AsyncHttpClient.cpp

DaeMonSxy, 2023-10-23:
Hi, jumping in also.

assert failed: tcp_update_rcv_ann_wnd IDF/components/lwip/lwip/src/core/tcp.c:951 (new_rcv_ann_wnd <= 0xffff)

my solution seems working within 1-2-3 hours, then suddenly crashing.
how did anyone solved the issue?

asynctcp code changes?
platformio build_flags ? 

any solution is welcome.

drmpf, 2023-12-17:
I have the same error 
calling line is tcp_recved(msg->pcb, msg->received); from AsyncTCP.cpp
Back trace is
```
Decoding stack results
0x40083dc5: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c line 408
0x4008bfa5: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c line 137
0x40091571: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c line 85
0x400fd18e: tcp_update_rcv_ann_wnd at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp.c line 951
0x400fd23c: tcp_recved at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp.c line 991
0x400e0dea: _tcp_recved_api(tcpip_api_call_data*) at C:\Users\matthew\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp line 422
0x400f9f24: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 172
```

Kamdzik, 2024-04-09:
hello I have exactly the same error , has anyone solved it yet?

DaeMonSxy, 2024-04-10:
Asynctcp has its own issues. 
Im also waiting for someone who would rewrite the lib, and share a proper/bug less version.

Meanwhile the only thing I have found is to change in the lib the line: 
`_async_queue = xQueueCreate(256, sizeof(lwip_event_packet_t *));  // dms/ prevents "async_tcp" Watchdog error `

original contains 32, 

Iam using that way, seems less prone to crash, hope it help, reply if so.