me-no-dev/AsyncTCP #91: Please write some examples

danielGheorghe, 2020-04-19:
Will be useful: async TCP Server, async TCP client and async TCP multicast server.
There is not one example on internet for how to use this library without one http server. No one know how to use it as that. I do want simple and fast tcp communication, i do not want example with "ESPAsyncWebServer". I do not need web server, only tcp communication.

bertmelis, 2020-04-19:
It really is easier than you think. Just look at the header files and you'll figure out.

- create async object
- attach callbacks
- connect
and there you go, you've got yourself and async TCP client.

On the server side it's almost the same. On every incoming connection you receive a client object. You can use this to attach your callbacks.

danielGheorghe, 2020-04-20:
Show me one example on all internet if is that simple. No, is not. Should have examples.

bertmelis, 2020-04-20:
I didn't check if it compiles.
You should understand how it works now.

PS I didn't include the code to setup WiFi.

```C++
#include <Arduino.h>
#include <AsyncTCP.h>

AsyncClient client;
constexpr IPAddress serverIP(192, 168, 1, 10);
constexpr uint16_t serverPort = 8888;

void onConnect(void* arg, AsyncClient* c) {
  Serial.print("We're connected!\n");
  c.write("hello server");
}

void onDisconnect(void* arg, AsyncClient* c) {
  Serial.print("We're disconnected!\n");
}

void onData(void* arg, AsyncClient* c, void* data, size_t len) {
  uint8_t* d = reinterpret_cast<uint8_t*>(data);
  Serial.print("We received some data: 0x");
  for (size_t i = 0; i < len; ++i) Serial.printf("%02x", d[i]);
  Serial.print("\n");
}

void setup() {
  Serial.begin(115200);
 client.onConnect(onConnect);
  client.onDisconnect(onDisconnect);
  client.onData(onData);

  // I've left out the WiFi connecting stuff.
  // Obviously, you can only call connect when WiFi is up.
  client.connect(serverIP, serverPort);
}

void loop() {
  delay(1);
}
```

danielGheorghe, 2020-04-30:
A, thanks a lot! Sorry for delay but your example is grate and should be in the project. Thanks again.

danielGheorghe, 2020-05-05:
I have to reopen that. I cannot figure out how to complete your example.
My Client code:
`
#include <Arduino.h>
#include <AsyncTCP.h>
#include "WiFi.h"

const char * ssid = "mySSID";
const char * password = "mypass";

AsyncClient client;

constexpr  IPAddress serverIP(192,168,1,118);
constexpr uint16_t serverPort = 8888;

void onConnect(void* arg, AsyncClient* c) {
  Serial.print("We're connected!\n");
  c.write("hello server");
}

void onDisconnect(void* arg, AsyncClient* c) {
  Serial.print("We're disconnected!\n");
}

void onData(void* arg, AsyncClient* c, void* data, size_t len) {
  uint8_t* d = reinterpret_cast<uint8_t*>(data);
  Serial.print("We received some data: 0x");
  for (size_t i = 0; i < len; ++i) Serial.printf("%02x", d[i]);
  Serial.print("\n");
}

void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.println("WiFi Failed");
        while(1) {
            delay(1000);
        }
    }
 client.onConnect(onConnect);
  client.onDisconnect(onDisconnect);
  client.onData(onData);

  // I've left out the WiFi connecting stuff.
  // Obviously, you can only call connect when WiFi is up.
  client.connect(serverIP, serverPort);
}

void loop() {
  delay(1);
}
`
I have few errors: 
/////////////////////////////
1:
 request for member 'write' in 'c', which is of pointer type 'AsyncClient*' (maybe you meant to use '->' ?)
   c.write("hello server");

2:
asyncTcpTest:10:44: error: the type 'const IPAddress' of constexpr variable 'serverIP' is not literal
 constexpr  IPAddress serverIP(192,168,1,118);
////////////////////////////////

First was fixed by that suggestion: c->write("hello server"); I am not sure if is ok....
The seccond one i can not  figure out.

And for server side.... I am stuck here:

`#include <Arduino.h>
#include <AsyncTCP.h>
#include "WiFi.h"

const char * ssid = "mySSID";
const char * password = "mypass";
constexpr uint16_t serverPort = 8888;

AsyncServer server(serverPort);

void setup() {
  // put your setup code here, to run once:
    Serial.begin(115200);
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.println("WiFi Failed");
        while (1) {
            delay(1000);
        }
    }
    /////

}

void loop() {
  // put your main code here, to run repeatedly:

}`

I have call the constructor with AsyncServer server(serverPort);, i have made wifi connection and now i should call void onClient(AcConnectHandler cb, void* arg); ? How to do that? Where i find that AcConnectHandler ? 

Can you help me again? Please.

bertmelis, 2020-05-05:
Yeah, I made some mistakes. c is a pointer and you can leave out constexpr.

Now, as I told you, as soon as a client connects, the server object gives you an AsyncClient object. You have to attach your callbacks onto the client.

If you don't understand this, then maybe you should read yourself into c++, OOP and try something easier first. And from then on think "event driven" when you do async programming.

danielGheorghe, 2020-05-05:
yes, i will have to learn more. any thing i try i got errors about tcpip_api_call:

Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:259:27: error: field 'call' has incomplete type 'tcpip_api_call'

     struct tcpip_api_call call; 

maybe someone will be happy to help. anyway this library should have examples

closed 2020-07-18 by stale[bot]