me-no-dev/AsyncTCP #94: AsyncTCP/ESPAsyncWebServer and ESP-IDF AsyncUDP lock up networking if used simultaneously

avillacis, 2020-05-14:
If a single Arduino ESP32 project has code that uses for different purposes both AsyncUDP (from espressif/arduino-esp32) and AsyncTCP (underlying ESPAsyncWebServer), the ESP32 stops accepting or generating network requests after about one hour of both libraries being used simultaneously.

I have a few ESP32 projects written with Arduino IDE that make extensive use of ESPAsyncWebServer. All of these projects also require NTP time synchronization for accurate timestamps. I have been using the NTPClient library until now for this. The issue with NTPClient is that it is based on WiFiUDP, which does not provide callbacks on incoming UDP traffic, and therefore sort of forces a synchronous operation in which NTPClient sends a packet, and then blocks waiting for the response from the NTP server.

I wrote my own code for querying the NTP server and listening for its responses using AsyncUDP as provided by the espressif/arduino-esp32 project, the one installed when downloading ESP32 support for Arduino IDE. When run by itself, it works correctly. However, when used alongside a web server using ESPAsyncWebServer and AsyncUDP, and then the browser requests a long-running data update using either Websockets or Server-Sent Events, the device runs for a while and then the network traffic stops completely. No websocket/SSE data, no response to page refresh, no ping at all. However, the serial port shows that the rest of the code remains active. At this point only a reboot appears to fix the issue.

If I use the blocking NTPClient API alongside ESPAsyncWebServer, the network communication appears rock-solid.

My homebrewed NTP client code used to invoke the UDP packet sending from within a timer callback. I removed that and implemented a mode that instead sends the packet using a function that is to be invoked from the main loop() function, but the lockup issue persists.

avillacis, 2020-05-14:
Brainstorming: both AsyncTCP and AsyncUDP create and use tasks to receive network traffic and invoke the low-level lwip functions that do the actual work. Are they interfering with each other? Would this be solved if the AsyncTCP project provided UDP support as well using the same task for both?

avillacis, 2020-07-16:
No commits at all here, so I suppose the bug is still present.

avillacis, 2020-07-27:
I have found this pull request for an AsyncUDP bug that breaks the network stack after continued reception of UDP packets, so this particular bug report might not be in AsyncTCP at all: https://github.com/espressif/arduino-esp32/pull/3290

thbl, 2020-08-03:
+1 @me-no-dev

avillacis, 2020-10-08:
Still present as far as I know.

closed 2020-12-22 by stale[bot]

wachidsusilo, 2021-01-25:
Hello @avillacis, i have the same issue here. Did you found a workaround?

ralphrmartin, 2022-01-12:
This may or may not be the same issue, but there is definitely some bad interaction here. Using AsyncTCP (for AsyncWebServer) together with the Espressif NTP code does not work properly - while the RTC is synced when the NTP code is initialised, the expected resync at regular intervals does not.