me-no-dev/AsyncTCP #97: assertion "invalid socket state for sent callback" failed.... function: tcp_sent

TryingToGITthis, 2020-06-24:
I'm seeing these errors on an ESP32 acting as an AP with multiple other ESP32 clients.  One client connects every 15 seconds and a couple others every minute or so.  If no clients connect, the code runs for months just fine. Only when clients connect does the AP throw an error similar to this one and the board reboots.

Is this an issue with the AsyncTCP library or deeper in the ESP-IDF?

Original error:
``
ssertion "invalid socket state for sent callback" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c", line 1842, function: tcp_sent
abort() was called at PC 0x401207c7 on core 1
``

Decoded:

``
0x40092a88: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
0x40092cb9: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707      
0x401207c7: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
0x4013aa92: tcp_sent at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c:1765
0x400e67f8: AsyncClient::_error(signed char) at C:\Users\Paul\Documents\Arduino\libraries\AsyncTCP-master\src/AsyncTCP.cpp:1065      
0x400e6a97: AsyncClient::_s_error(void*, signed char) at C:\Users\Paul\Documents\Arduino\libraries\AsyncTCP-master\src/AsyncTCP.cpp:1065
  \-> inlined by: _handle_async_event at C:\Users\Paul\Documents\Arduino\libraries\AsyncTCP-master\src/AsyncTCP.cpp:174
  \-> inlined by: _async_service_task at C:\Users\Paul\Documents\Arduino\libraries\AsyncTCP-master\src/AsyncTCP.cpp:197
0x4008f19d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
``

me-no-dev, 2020-06-25:
No this is deeper than Async. LwIP got an ACK for a closed connection or something like that. Really weird. Mayabe keeps some connections and new clients get the same IP of the old ones?

TryingToGITthis, 2020-06-25:
Swapping to AsyncUDP resolved all the issues. I had issues with DHCP and SoftAP so I gave each client a unique IP. Other issues I was seeing were connection issues, ACK timeout issues, etc. It just seems like if you have multiple devices constantly connecting to the ESP32 via TCP, it has issues.

closed 2020-09-08 by stale[bot]

HamzaHajeir, 2023-04-29:
Any further information to this error? I'm facing it even with another library (as it's sourced from lwip tcp)