me-no-dev/AsyncTCP #109: onConnect doesn't fire any more

daduke, 2020-11-01:
Hi,

I've been using AsynTCP in this project: https://github.com/jamct/DoorsignEPD for two years without a problem. Now I wanted to implement a new feature, but the onConnect callback doesn't fire any more. I presume it is caused by an update sometime between 2018 and now, but I can't tell whether it's the Arduino IDE, the ESP32 tool chain or one of the libraries involved. I do get a wifi connection, but either connect() doesn't work or the callback doesn't get called. Any ideas? When I flash back the binary I compiled from the same code 2 years ago, everything works...

thanks,
-Christian

daduke, 2020-11-02:
diving into the AsyncTCP code,
`int8_t AsyncClient::_connected(void* pcb, int8_t err){`
never gets called...

BlueAndi, 2020-11-02:
I am using AsyncTcp here too and its working:
https://github.com/BlueAndi/esp-rgb-led-matrix/blob/28a28b034e19350ac1a94a10937273c8ba0f5074/src/Web/AsyncHttpClient.cpp#L91-L124

A stupid question, but you are sure you are connected to wifi? :-)

daduke, 2020-11-02:
> A stupid question, but you are sure you are connected to wifi? :-)
pretty much, yeah. My DHCP server hands out an address and `WiFi.status() == WL_CONNECTED`.

> Any other callback like onError() firing?
not that I can tell. It's like I call `client.connect(host, httpPort)` and that's the last thing I hear of AsyncTCP...

BlueAndi, 2020-11-02:
And you get every 2s "Connection setup in progres" via serial interface, right?
Can you additional print something in case the wifi connection is lost?

daduke, 2020-11-02:
I have a lead now: the original project uses https://github.com/ct-Open-Source/Basecamp for wifi setup but this causes memory issues for larger epaper displays. I removed Basecamp and handled wifi setup via WiFi.h. In the past this used to work, but now apparently WiFi.h doesn't play along with AsyncTCP. I'll try to handle wifi differently. Thanks for listening tho :)

BlueAndi, 2020-11-02:
You are welcome. :-)

closed 2021-01-16 by stale[bot]