me-no-dev/AsyncTCP #111: AsyncTCP.cpp: Double-Free in event loop in _handle_async_event() when e->event == LWIP_TCP_CLEAR

ul-gh, 2020-11-16:
Hi,

in AsyncTCP.cpp in function _handle_async_event(lwip_event_packet_t * e), when the event is a TCP_CLEAR, the handler calls _remove_events_with_arg(e->arg) which not only removes the event packet pointer from the _async_queue but also frees the memory adressed by the packet pointer:

```
static bool _remove_events_with_arg(void * arg){
    lwip_event_packet_t * first_packet = NULL;
    [...]
    while(!first_packet){
        if(xQueueReceive(_async_queue, &first_packet, 0) != pdPASS){
            return false;
        }
        //discard packet if matching
        if((int)first_packet->arg == (int)arg){
            free(first_packet);  <=========== This is a lwip_event_packet_t*
            first_packet = NULL;
        [...]
    }
```

However, when this returns back to _handle_async_event(lwip_event_packet_t * e), then at the end of the large if...else if...else statement, another call is made to stdlib free():

```
static void _handle_async_event(lwip_event_packet_t * e){
    if(e->arg == NULL){
        // do nothing when arg is NULL
        //ets_printf("event arg == NULL: 0x%08x\n", e->recv.pcb);
    } else if(e->event == LWIP_TCP_CLEAR){
        _remove_events_with_arg(e->arg);
    } else if {
       [....]
    }
    free((void*)(e)); (sic) <=========== The same event_packet_t*
}
```

IIRC, this is a double-free issue with all its implications...

I realise the _async_queue itself is call/return-by-value, however it writes to a ptr-to-ptr buffer and the memory freed again then afterwards in fact is the same which was freed before.

Please excuse if I am wrong.

The easiest fix then is adding a return() from the handler function after the _remove_events_with_arg(e->arg):
```
    } else if(e->event == LWIP_TCP_CLEAR){
        _remove_events_with_arg(e->arg);
        return();
    } else if {
       [....]
    }
    free(e);
}
```

ul-gh, 2020-11-17:
I have to add that the double-free can/could only happen if there are/were duplicate events (i.e. identical copies of the same pointer) in the queue - I am not sure if this can happen.

Because at the time when _remove_events_with_arg() pulls events out of the _async_queue, the original LWIP_TCP_CLEAR type event had already been pulled out of it before and so this can only be the same pointer if there are duplicates in the queue.

So, rewording the title as a question ;-)

Of course, when we change the signature of _remove_events_with_arg(e->arg) such that it is called not with the arg but with the event packet pointer itself, then we can play it safe and free the event inside the function. Then, after NULL-ifying any copies of it which could be in the event queue, a final and possible double call to free() on the nullified copy would be safe in any case.

BlueAndi, 2020-11-18:
Hmm ... I don't see the double free for the same element, because _remove_events_with_arg() is working only the elements in the buffer, but the one which is handled by _handle_async_event() isn't in the buffer anymore.

And isn't the idea in case of LWIP_TCP_CLEAR to remove only the event, related to the same AsyncClient instance?

ul-gh, 2020-11-19:
Hi, you are right, I had it wrong when I first looked at this.

I went through the code again, the event package pointer is never copied but only assigned to by malloc(), so there are no duplicate events. Closing this now...

closed 2020-11-19 by ul-gh