me-no-dev/AsyncTCP #114: FR: Allow configuring LwIP's keepalives

positron96, 2021-02-23:
Hi. 
Is it possible to add support for TCP keepalives (and methods to enable/disable/configure them)? As I understand, LwIP has this functionality and it can be enabled on per-socket basis (I just don't know if it's compiled into ESP32 core). 

You see, it's not uncommon in embedded world to just unplug the device from power, in this case a TCP connection will be hanging on the other side and some non-trivial logic would be required to detect and close it. I am testing an application with AsyncServer on one ESP32 and long-lasting TCP connection to WiFiClient on other ESP32, and this ESP32 can be unplugged from power without proper termination. In this case I don't see a `onDisconnect` called on server side, as well as no events (timeouts, errors) that could help in detecting broken connection. The only approach at the moment is to write my own keep-alive logic, but that would be so cumbersome, not to mention changes in upper-level protocol.

Might be useful: https://stackoverflow.com/questions/29492995/lwip-stack-netconn-api-keep-connection-keep-alive

copercini, 2021-06-02:
It's a bit hacky, but you can add keepalive adding these lines after here: https://github.com/me-no-dev/AsyncTCP/blob/ca8ac5f919d02bea07b474531981ddbfd64de97c/src/AsyncTCP.cpp#L1318

```
            pcb->so_options |= SOF_KEEPALIVE; //Turn on TCP Keepalive for the given pcb
            // Set the time between keepalive messages in milli-seconds
            pcb->keep_idle = 45000; // 45 seconds
            pcb->keep_intvl = 45000; // 45 seconds
            pcb->keep_cnt = 7; //The number of unanswered probes required to force closure of the socket
            // printf("keep alive is on"); //low level debug only
```

positron96, 2021-06-03:
@copercini big thanks, I'll try to make my own fork with this!

closed 2021-08-17 by stale[bot]