me-no-dev/AsyncTCP #119: Help to interpretet  "[E][AsyncTCP.cpp:729] connect(): error: -16 "

lgLindstrom, 2021-05-21:
Hi

I am using  AsyncMqttClient  wich is using AsyncTCP. When  AsyncMqttClient  tries to connect to a Mqtt Broker I get this message from AsyncTCP.

`[E][AsyncTCP.cpp:729] connect(): error: -16 `

I have found no way to understand what error: -16 means.

My application is a VsCode/PlatformIO Arduino based project. 
The Mqtt Broker is working nicely,,, i.e I can connect to it from other clients. 
Wifi is properly setup  in my application i.e the application can do other stuff over Wifi .

My development board is DevkitC from Espressif.

Can anyone give some help here :)

BlueAndi, 2021-06-13:
I guess:
https://github.com/espressif/esp-lwip/blob/2195f7416fb3136831babf3e96c027a73075bd4f/src/include/lwip/err.h#L88

closed 2021-08-28 by stale[bot]

razdavidovich, 2022-07-10:
I have the same issue, has anyone figured this out ?

HamzaHajeir, 2022-12-13:
Hi there,
I've found that I've passed an empty url to connect().
Also, if you're trying to pass an IP Address as text, the library doesn't support that. 
checkout https://github.com/philbowles/AardvarkTCP/blob/master/src/H4AsyncTCP.cpp#L267