me-no-dev/AsyncTCP #121: Sending big blocks of data triggers async_tcp task watchdog

detmot5, 2021-08-04:
Hey,
When serving larger files (like 250KB) i got problem with triggering wdt in async_tcp task. After disabling it website still failed to load.
After some research I found source of the problem - too small size of _async_queue. Changing it to 256 or 512 solves my issue.

``` c++
static inline bool _init_async_event_queue(){
    if(!_async_queue){
        _async_queue = xQueueCreate(512, sizeof(lwip_event_packet_t *));
        if(!_async_queue){
            return false;
        }
    }
    return true;
}
```
It worked for me.

nietaki, 2021-09-29:
Can confirm this helps for me as well - left a comment on the PR how to use the changed code from a platformio project.

proddy, 2021-11-06:
And this also fixed my issues when serving large files via ESPAsyncWebServer - so many thanks for posting this @detmot5 !

detmot5, 2021-11-06:
> Can confirm this helps for me as well - left a comment on the PR how to use the changed code from a platformio project.

> Can confirm this helps for me as well - left a comment on the PR how to use the changed code from a platformio project.

Hey, you should clone this library into "lib" folder and change it here. Or fork it and pass it into lib_deps as github link

zekageri, 2021-11-08:
Where is `_async_queue` defined?

proddy, 2021-11-08:
> Where is `_async_queue` defined?

what do you mean? There's only one .cpp file ;)

zekageri, 2021-11-08:
Oh. I got a tweaked version of the [AsyncTCP ](https://github.com/yubox-node-org/AsyncTCPSock) lib by @yubox-node-org.
There is no ` _async_queue ` in there. :D

closed 2022-04-16 by stale[bot]

infwingit, 2022-10-16:
You helped me a lot , BIG THANKS !!!