me-no-dev/AsyncTCP #123: Task watchdog got triggered. The following tasks did not reset the watchdog in time: - async_tcp (CPU 1)

brandonros, 2021-08-16:
```
E (77747) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (77747) task_wdt:  - async_tcp (CPU 1)
E (77747) task_wdt: Tasks currently running:
E (77747) task_wdt: CPU 0: IDLE
E (77747) task_wdt: CPU 1: async_tcp
E (77747) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x400E897B:0x3FFB06500x400821F9:0x3FFB0670 0x4014CE6B:0x3FFBB1C0 0x400D280A:0x3FFBB1E0 0x400887E5:0x3FFBB200 0x4008B72D:0x3FFBB220 
0x400e897b: task_wdt_isr at /Users/brandonros/Desktop/esp-idf-v4.4/components/esp_system/task_wdt.c:191 (discriminator 3)

0x400821f9: _xt_lowint1 at /Users/brandonros/Desktop/esp-idf-v4.4/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x4014ce6b: cpu_ll_waiti at /Users/brandonros/Desktop/esp-idf-v4.4/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at /Users/brandonros/Desktop/esp-idf-v4.4/components/esp_pm/pm_impl.c:841

0x400d280a: esp_vApplicationIdleHook at /Users/brandonros/Desktop/esp-idf-v4.4/components/esp_system/freertos_hooks.c:63

0x400887e5: prvIdleTask at /Users/brandonros/Desktop/esp-idf-v4.4/components/freertos/tasks.c:3823 (discriminator 1)

0x4008b72d: vPortTaskWrapper at /Users/brandonros/Desktop/esp-idf-v4.4/components/freertos/port/xtensa/port.c:168

E (77747) task_wdt: Print CPU 1 backtrace

Backtrace:0x400848BD:0x3FFB0C500x400821F9:0x3FFB0C70 0x4008A1B0:0x3FFCCC20 0x400E3077:0x3FFCCC40 0x400F4816:0x3FFCCC60 0x400E9516:0x3FFCCC80 0x400E954D:0x3FFCCCC0 0x4014D057:0x3FFCCCE0 0x400EEE81:0x3FFCCD00 0x400EEEA2:0x3FFCCD50 0x400EF595:0x3FFCCD80 0x400EF5B5:0x3FFCCDC0 0x40143D0D:0x3FFCCDE0 0x40143D5D:0x3FFCCE10 0x40143DA1:0x3FFCCE30 0x40143EBE:0x3FFCCE50 0x40143F8A:0x3FFCCE70 0x4008B72D:0x3FFCCEA0 
0x400848bd: esp_crosscore_isr at /Users/brandonros/Desktop/esp-idf-v4.4/components/esp_system/crosscore_int.c:92

0x400821f9: _xt_lowint1 at /Users/brandonros/Desktop/esp-idf-v4.4/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x4008a1b0: _frxt_dispatch at /Users/brandonros/Desktop/esp-idf-v4.4/components/freertos/port/xtensa/portasm.S:461

0x400e3077: __yield at /Users/brandonros/Desktop/esp32-bench-flasher/components/arduino-esp32/cores/esp32/esp32-hal-misc.c:78

0x400f4816: ESP32SJA1000Class::endPacket() at /Users/brandonros/Desktop/esp32-bench-flasher/components/arduino-CAN/src/ESP32SJA1000.cpp:173

0x400e9516: can_send(unsigned int, unsigned char*, unsigned int) at /Users/brandonros/Desktop/esp32-bench-flasher/main/main.cpp:57

0x400e954d: on_socket_event(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int) at /Users/brandonros/Desktop/esp32-bench-flasher/main/main.cpp:162

0x4014d057: std::_Function_handler<void (AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int), void (*)(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int)>::_M_invoke(std::_Any_data const&, AsyncWebSocket*&&, AsyncWebSocketClient*&&, AwsEventType&&, void*&&, unsigned char*&&, unsigned int&&) at /Users/brandonros/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:297

0x400eee81: std::function<void (AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int)>::operator()(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int) const at /Users/brandonros/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687

0x400eeea2: AsyncWebSocket::_handleEvent(AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int) at /Users/brandonros/Desktop/esp32-bench-flasher/components/ESPAsyncWebServer/src/AsyncWebSocket.cpp:932

0x400ef595: AsyncWebSocketClient::_onData(void*, unsigned int) at /Users/brandonros/Desktop/esp32-bench-flasher/components/ESPAsyncWebServer/src/AsyncWebSocket.cpp:752

0x400ef5b5: std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebSocketClient::AsyncWebSocketClient(AsyncWebServerRequest*, AsyncWebSocket*)::{lambda(void*, AsyncClient*, void*, unsigned int)#7}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at /Users/brandonros/Desktop/esp32-bench-flasher/components/ESPAsyncWebServer/src/AsyncWebSocket.cpp:485
 (inlined by) _M_invoke at /Users/brandonros/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:297

0x40143d0d: std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at /Users/brandonros/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687

0x40143d5d: AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at /Users/brandonros/Desktop/esp32-bench-flasher/components/AsyncTCP/src/AsyncTCP.cpp:934

0x40143da1: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at /Users/brandonros/Desktop/esp32-bench-flasher/components/AsyncTCP/src/AsyncTCP.cpp:1210

0x40143ebe: _handle_async_event(lwip_event_packet_t*) at /Users/brandonros/Desktop/esp32-bench-flasher/components/AsyncTCP/src/AsyncTCP.cpp:162

0x40143f8a: _async_service_task(void*) at /Users/brandonros/Desktop/esp32-bench-flasher/components/AsyncTCP/src/AsyncTCP.cpp:197

0x4008b72d: vPortTaskWrapper at /Users/brandonros/Desktop/esp-idf-v4.4/components/freertos/port/xtensa/port.c:168
```

Snippet:

```
#include <Arduino.h>
#include <WiFi.h>
#include <CAN.h> // https://github.com/sandeepmistry/arduino-CAN
#include <AsyncTCP.h> // https://github.com/me-no-dev/AsyncTCP
[... 158 lines cut ...]
  }
}

void setup() {
  // serial
  Serial.begin(115200);
  // can
  // can_init(500000);
  // wifi
  WiFi.softAP("test", "test123");
  Serial.printf("IP: %s\n", WiFi.softAPIP().toString().c_str());
  // web server
  server.on("/", HTTP_GET, on_index);
  server.onNotFound(on_not_found);
  // web socket
  ws.onEvent(on_socket_event);
  server.addHandler(&ws);
  // start
  server.begin();
}

void loop() {
  vTaskDelay(10);
}
```

bertmelis, 2021-08-17:
You cannot use 'yield' or 'delay' in the async callbacks.

brandonros, 2021-08-18:
can you vTaskDelay?

bertmelis, 2021-08-18:
> can you vTaskDelay?

Same thing. You cannot use any form of blocking code.

A way of solving this is to set a flag (global boolean var?) and act upon the flag in your `void loop()`.

positron96, 2021-11-24:
Hi. 
I am having the same issue. Though it seems like in my case the cause is not `delay`/`yeild`, but 30 lines of `ets_printf` from async handler. `ets_printf` prints to UART at 115200 and it's probably too slow for watchdog. However, if I replace `ets_printf` with ordinary `Serial.println`, the watchdog does not kick in. Is logging also prohibited inside AsyncTCP handlers? 

Can someone provide more definitive info about this issue? Are AsyncTCP handlers run from ISRs? If handlers are run from normal task, then delays would be fine inside them. Is there some kind of `esp_task_wdt_reset` that I can call to feed the watchdog?

closed 2022-04-16 by stale[bot]