me-no-dev/AsyncTCP #138: Is there a good async replacement for HTTPClient?

benpeart, 2022-01-18:
I can't figure out a way to send you a question - hopefully this will work instead. :)

I have a kaleidoscope project where I'm using ESPAsyncWebServer but also need to make REST calls out to external APIs. I have not been able to find a library in your Async family that I can use.

For now, I've written the REST calls using the standard HTTPClient.h file to do my HTTP GET requests but when I included that file, my project grew by 179K. It also has made my project unstable and now it crashes randomly after running for several hours. I'm not 100% sure that is the cause but I wonder if there are conflicts between HTTPClient.h and the Async* libraries running simultaneously.

Is there a better HTTPClient library I can use that won't grow my code size so much and won't destabilize my project? I looked at the AsyncClient class in your AsyncTCP library but it isn't as user friendly as HTTPClient.h - I'm not sure how to make simple REST calls using it. Is that a reasonable use for it? Do you have any documentation or example of how I would do that? Any other suggestions?

Thanks,

Ben

bertmelis, 2022-01-19:
a TCP client is not a HTTP client, but is built on top of that.

check this out: https://github.com/boblemaire/asyncHTTPrequest

benpeart, 2022-01-26:
> a TCP client is not a HTTP client, but is built on top of that.
> 
> check this out: https://github.com/boblemaire/asyncHTTPrequest

For anyone else who comes across this, after debugging some issues with asyncHTTPrequest, I found https://github.com/khoih-prog/AsyncHTTPRequest_Generic which is an updated fork of Bob's library.

closed 2022-01-26 by benpeart