me-no-dev/AsyncTCP #147: Arduino Library Registry

r41d, 2022-05-16:
Hello there :)

Have you considered submitting this library to the official Arduino Library Registry? (https://github.com/arduino/library-registry)

That way, this library could easily be installed in the Arduino IDE via the built-in Library Manager.

The same holds true for ESPAsyncTCP and ESPAsyncWebServer of course :)

r41d, 2022-05-16:
As far as I can tell everything in the `library.properties` files of these three libraries already looks great, the only thing that would be necessary would be the a PR on the registry repo. The process seems to be rather quick

r41d, 2022-08-01:
Bad bot! Don't just close issues when no author has said anything yet

r41d, 2022-09-28:
So, no intention of including this lib in the Arduino package registry?

roblatour, 2022-10-28:
I would love to have it added into the the  Arduino package registry as well

per1234, 2023-05-22:
Due to the lack of action/interest from @me-no-dev in registering the library in Arduino Library Manager, a community member forked this repository and the other two related libraries and registered the forks:

- https://github.com/arduino/library-registry/pull/2233
- https://github.com/arduino/library-registry/pull/2235

So I think it is fine to let the bot have its way and close the issue.

### Related

- https://github.com/me-no-dev/ESPAsyncWebServer/issues/468
- https://github.com/me-no-dev/ESPAsyncTCP/issues/139