me-no-dev/AsyncTCP #174: .pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp:85:8: : 'xQueueHandle' does not name a type;

y9Kap, 2023-12-24:
[image]
Error compiling AsyncMqttClient

y9Kap, 2023-12-24:
Platformio.ini: 
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
monitor_speed = 115200

ForrestFire0, 2024-01-30:
Also having this issue. I tried the stable and latest arduino-esp32.

Incredibly, just changing ```xQueueHandle``` to ```QueueHandle_t``` allowed me to compile.

I note that your platform.ini file does not include the library listed...

krzychoooo, 2024-06-11:
Hello
my platformio.ini file:
```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
upload_port = COM6
lib_deps = me-no-dev/ESP Async WebServer@^1.2.4
```
Compilation result:
```

 *  Executing task in folder WebService: C:\Users\fajny\.platformio\penv\Scripts\platformio.exe run 

Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (2024.4.14) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, 
olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.0.0+sha.08e138f
 - tool-esptoolpy @ 4.7.2
 - tool-mklittlefs @ 3.2.0
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP Async WebServer @ 1.2.4
|-- AsyncTCP @ 1.1.1
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio\build\esp32doit-devkit-v1\src\main.cpp.o
Building .pio\build\esp32doit-devkit-v1\bootloader.bin
Generating partitions .pio\build\esp32doit-devkit-v1\partitions.bin
esptool.py v4.7.2
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio\build\esp32doit-devkit-v1\lib4d4\AsyncTCP\AsyncTCP.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib75b\FS\FS.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib75b\FS\vfs_api.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\libf7d\Network\NetworkClient.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\libf7d\Network\NetworkEvents.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\libf7d\Network\NetworkInterface.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\libf7d\Network\NetworkManager.cpp.o
.pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp:79:8: error: 'xQueueHandle' does not name a type; did you mean 'QueueHandle_t'?
   79 | static xQueueHandle _async_queue;
      |        ^~~~~~~~~~~~
      |        QueueHandle_t
.pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp: In function 'bool _init_async_event_queue()':
.pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp:97:9: error: '_async_queue' was not declared in this scope
   97 |     if(!_async_queue){
      |         ^~~~~~~~~~~~
.pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp: In function 'bool _send_async_event(lwip_event_packet_t**)':
.pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp:107:12: error: '_async_queue' was not declared in this scope
  107 |     return _async_queue && xQueueSend(_async_queue, e, portMAX_DELAY) == pdPASS;
      |            ^~~~~~~~~~~~
.pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp: In function 'bool _prepend_async_event(lwip_event_packet_t**)':
.pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp:111:12: error: '_async_queue' was not declared in this scope
  111 |     return _async_queue && xQueueSendToFront(_async_queue, e, portMAX_DELAY) == pdPASS;
      |            ^~~~~~~~~~~~
.pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp: In function 'bool _get_async_event(lwip_event_packet_t**)':
.pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp:115:12: error: '_async_queue' was not declared in this scope
  115 |     return _async_queue && xQueueReceive(_async_queue, e, portMAX_DELAY) == pdPASS;
      |            ^~~~~~~~~~~~
.pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp: In function 'bool _remove_events_with_arg(void*)':
.pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp:122:9: error: '_async_queue' was not declared in this scope
  122 |     if(!_async_queue){
      |         ^~~~~~~~~~~~
.pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp:127:26: error: '_async_queue' was not declared in this scope
  127 |         if(xQueueReceive(_async_queue, &first_packet, 0) != pdPASS){
      |                          ^~~~~~~~~~~~
In file included from C:/Users/fajny/.platformio/packages/framework-arduinoespressif32/tools/esp32-arduino-libs/esp32/include/freertos/FreeRTOS-Kernel/include/freertos/semphr.h:42,
                 from C:/Users/fajny/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:35,
                 from .pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp:22:
.pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp:135:30: error: '_async_queue' was not declared in this scope
  135 |         } else if(xQueueSend(_async_queue, &first_packet, portMAX_DELAY) != pdPASS){
      |                              ^~~~~~~~~~~~
C:/Users/fajny/.platformio/packages/framework-arduinoespressif32/tools/esp32-arduino-libs/esp32/include/freertos/FreeRTOS-Kernel/include/freertos/queue.h:545:26: note: 
in definition of macro 'xQueueSend'
  545 |     xQueueGenericSend( ( xQueue ), ( pvItemToQueue ), ( xTicksToWait ), queueSEND_TO_BACK )
      |                          ^~~~~~
.pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp:140:22: error: '_async_queue' was not declared in this scope
  140 |     while(xQueuePeek(_async_queue, &packet, 0) == pdPASS && packet != first_packet){
      |                      ^~~~~~~~~~~~
*** [.pio\build\esp32doit-devkit-v1\lib4d4\AsyncTCP\AsyncTCP.cpp.o] Error 1
```
Proszę daj mi jakąś pomoc.

tiecia, 2024-09-01:
To anyone having this issue with PlatformIO, the library hasn't been updated in the PIO registry with the commit that fixes this 58cbe1fabe78977e3140391c01c03f0cb51e347c.

In `platformio.ini` you must replace `lib_deps = me-no-dev/ESPAsyncTCP` with `lib_deps = https://github.com/me-no-dev/AsyncTCP`. This fetches the latest commit from the repo directly instead of the commit published to the registry.

Even better, you can do this to fetch a specific commit instead of fetching the master branch. `lib_deps = https://github.com/me-no-dev/AsyncTCP#17039c38d2c6a6e39db007b011d8730dc1e6585d`. This just points to the latest commit as of writing this comment.