me-no-dev/AsyncTCP #178: fatal error: sdkconfig.h: No such file or directory

stanislav-server, 2024-03-01:
In file included from C:\Users\Станислав\Documents\Arduino\Lan_server__PVZ\Lan_server__PVZ.ino:3:
c:\Users\���������\Documents\Arduino\libraries\AsyncTCP\src/AsyncTCP.h:26:10: fatal error: sdkconfig.h: No such file or directory
   26 | #include "sdkconfig.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1

castruck, 2024-07-11:
Brand new install of Arduino IDE 2.3.2 using ESP8266 NodeMCU 1.0 (ESP 12-E Module) trying to create a web server instance but getting compilation error.

// Create an instance of the web server
ESPAsyncWebSrv server(80);

In file included from C:\Users\castr\OneDrive\Documents\Arduino\arduino_ide_esp8266_nodemcu_pump_code_Copilot\arduino_ide_esp8266_nodemcu_pump_code_Copilot.ino:2:
c:\Users\castr\OneDrive\Documents\Arduino\libraries\AsyncTCP\src/AsyncTCP.h:26:10: fatal error: sdkconfig.h: No such file or directory
   26 | #include "sdkconfig.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1

me-no-dev, 2024-07-12:
`AsyncTCP` is for ESP32. For ESP8266 you need `ESPAsyncTCP`