me-no-dev/AsyncTCP #179: Problem when connecting via WIFIAP

butaikis, 2024-03-06:
Hi @me-no-dev , there is a problem when connecting to ESP32 via WiFiAp, the page does not load. It freezes constantly and then reboots. I tried to turn off the watchdog timer, but it didn't do anything. The problem is with WIFIAp, when accessed via ETH, WI-FI STA, everything loads well. Below are the last lines of the log. Maybe there is a way to solve it?

```
V (34276) esp_littlefs: ------------------------ Sem Taking [async_tcp]␛[0m
V (34283) esp_littlefs: --------------------->>> Sem Taken [async_tcp]␛[0m
V (34290) esp_littlefs: ---------------------<<< Sem Give [async_tcp]␛[0m
V (34296) esp_littlefs: ------------------------ Sem Taking [async_tcp]␛[0m
V (34303) esp_littlefs: --------------------->>> Sem Taken [async_tcp]␛[0m
V (34309) esp_littlefs: ---------------------<<< Sem Give [async_tcp]␛[0m
V (34316) esp_littlefs: ------------------------ Sem Taking [async_tcp]␛[0m
V (34323) esp_littlefs: --------------------->>> Sem Taken [async_tcp]␛[0m
V (34329) esp_littlefs: ---------------------<<< Sem Give [async_tcp]␛[0m
V (34336) esp_littlefs: ------------------------ Sem Taking [async_tcp]␛[0m
V (34342) esp_littlefs: --------------------->>> Sem Taken [async_tcp]␛[0m
V (34349) esp_littlefs: ---------------------<<< Sem Give [async_tcp]␛[0m
␛[0;31mE (43394) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:␛[0m
␛[0;31mE (43394) task_wdt:  - async_tcp (CPU 0/1)␛[0m
␛[0;31mE (43394) task_wdt: Tasks currently running:␛[0m
␛[0;31mE (43394) task_wdt: CPU 0: IDLE␛[0m
␛[0;31mE (43394) task_wdt: CPU 1: IDLE␛[0m
␛[0;31mE (43394) task_wdt: Aborting.␛[0m

abort() was called at PC 0x4015c8d8 on core 0
```

butaikis, 2024-03-20:
The problem is solved by changing the file
https://github.com/me-no-dev/AsyncTCP/blob/ca8ac5f919d02bea07b474531981ddbfd64de97c/src/AsyncTCP.cpp#L98

`change 32 to 64`

closed 2024-03-20 by butaikis

stevesch, 2024-03-23:
Where are these fixes going?  I don’t see any recent commits in the main
repo.